SpenceKonde/ATTinyCore #106: error: 'TWBR' was not declared in this scope

Genoil, 2017-01-26:
I'm trying to use a library that depends on the Wire library, but I can't get around the following error:

/Users/*****/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.2/libraries/Wire/Wire.cpp:82:3: error: 'TWBR' was not declared in this scope

How can I fix this?

SpenceKonde, 2017-01-26:
You need to use an I2C library that is compatible with the tiny84 - The stock Wire library depends on the chip having hardware I2C support, while the Tiny84 only has a USI. The USI can be used to do I2C, but it needs a different library - https://github.com/SpenceKonde/TinyWireM (assuming you're using it as an I2C master)

Unfortunately, you'll also need to modify the library you're using to include TinyWireM  instead of Wire, and possibly make other changes to the library to account for differences between TinyWireM and Wire. 

My plan is to create versions of SPI and Wire libraries to include with the core that will detect the appropriate implementation of I2C (or SPI) for the chip hardware, and present a unified API that will "just work" with existing libraries. The issue for that is here - https://github.com/SpenceKonde/ATTinyCore/issues/53 - I just haven't had time to implement it yet, as it's a rather big deal. I am acutely aware of how important this feature is to people.

closed 2017-01-26 by SpenceKonde

Genoil, 2017-01-26:
Thanks. Our prototype was using a Trinket (tiny85) which was working with their port of Wire, but I couldn't figure out to make it work with my tiny84. Ultimately fixed it with with some global find and replace Wire > TinyWireM in the I2C device libraries. Sorry to bother you.

SpenceKonde, 2017-01-26:
Glad to hear you got it sorted!

As I said, I realize this process kinda sucks. I'm working on the better solution for the future.

manix, 2023-11-25:
I am getting this error when I try to compile my code for an attiny24. When I go to the lib suggested in comment 2 I see that I should not install it and that Wire.h should be supported, I'm using ATTinyCore 2.0

How can I resolve this issue?

Thanks!