SpenceKonde/ATTinyCore #107: ATtiny841 interrupt vector names

stuartsnow, 2017-01-30:
Trying to get to grips with using AVR timers directly and came across this one:

I looked up the name of the ATtiny841 timer1 compare A interrupt; the datasheet lists it as TIM1_COMPA (page 49). So I declare
```cpp
ISR(TIM1_COMPA_vect)
{
  PORTA ^= 0x80;        // Toggle PA7
}
```
but my pin is not toggled. Changing the name to `TIMER1_COMPA_vect` causes it to work as intended.
I will say I'm not 100 % on how ISRs work with Arduino but `TIMER1_COMPA_vect` seems to be the ISR name for the 328P; they changed it slightly for the 841. Is this deliberate? Compatibility?
I have also noticed there seems to be no checking on the ISR name, I can write whatever I like there and it doesn't cause an error if it's not right. A limitation of how Arduino handles defining ISRs?

SpenceKonde, 2017-01-30:
Here's what makes this even weirder. From core_timers.c in the core...

`

#if defined( TIM0_COMPA_vect ) && ! defined( TIMER0_COMPA_vect )
#define TIMER0_COMPA_vect TIM0_COMPA_vect
#endif

#if defined( TIM0_COMPB_vect ) && ! defined( TIMER0_COMPB_vect )
#define TIMER0_COMPB_vect TIM0_COMPB_vect
#endif

#if defined( TIM0_OVF_vect ) && ! defined( TIMER0_OVF_vect )
#define TIMER0_OVF_vect TIM0_OVF_vect
#endif

#if defined( TIM1_COMPA_vect ) && ! defined( TIMER1_COMPA_vect )
#define TIMER1_COMPA_vect TIM1_COMPA_vect
#endif

#if defined( TIM1_COMPB_vect ) && ! defined( TIMER1_COMPB_vect )
#define TIMER1_COMPB_vect TIM1_COMPB_vect
#endif

#if defined( TIM1_OVF_vect ) && ! defined( TIMER1_OVF_vect )
#define TIMER1_OVF_vect TIM1_OVF_vect
#endif
`

SpenceKonde, 2017-01-31:
OH! They did it one way, but not the other way!

SpenceKonde, 2017-01-31:
If you could let me know if this fixes things for you that would be much appreciated.

closed 2017-01-31 by SpenceKonde

stuartsnow, 2017-01-31:
OK I downloaded the master (twice!) and tried it, it doesn't seem to have worked for me.

This sketch works:
```cpp
// Sets up Timer1 CTC to give 1 s interrupt

void setup()
{
  DDRA |= 0x80;           // Set PA7 as output
  InitTimer1Int();
}

void loop()
{
  
}

void InitTimer1Int(void)
{
  cli();
  TCCR1A = 0;                 // No compare outputs, CTC mode
  TCCR1B = 0b00001000;        // CTC mode (OCR1A), stop timer
  TCNT1H = 0;                 // Zero the timer
  TCNT1L = 0;
  GTCCR = 0x01;               // Reset the prescaler
  OCR1AH = 0x7A;              // Set compare value to 0x7A12 (31250)
  OCR1AL = 0x12;
  TIMSK1 = 0b000000010;       // Interrupt on Compare Match A

  TCCR1B |= 0b00000100;        // Start timer, CTC mode, prescaler clk/256
  sei();
}

ISR(TIMER1_COMPA_vect)
{
  PORTA ^= 0x80;        // Toggle PA7
}
```

But if I change `TIMER1_COMPA_vect` to 'TIM1_COMPA_vect` my LED stops blinking.

oqibidipo, 2017-02-01:
You need `#include <core_timers.h>`

stuartsnow, 2017-02-01:
Gotcha. Works now, thanks for the support!

commit a8f60b35 (2017-01-31): Should fix #107

Cleaning up the mess atmel made of vector names - it now goes in both directions, and covers timer2.