SpenceKonde/ATTinyCore #110: ATtiny_1634.md diagram has MOSI/MISO swapped

cardre, 2017-02-15:
Digram on ATTinyCore/avr/extras/ATtiny_1634.md shows MOSI/MISO swapped. Diagram needs to be corrected.

SpenceKonde, 2017-02-15:
Huh! Well this is actually considerably more complicated...

The USI and SPI programming pins are the same pins - presumably using the same underlying serial interface hardware. However, unlike "real" SPI, USI has DI and DO, not MISO and MOSI. Thus, the input and output lines for ISP programming are opposite what would be used when using the USI as an SPI master. 

I'm not sure how we want to handle this. I think ideally we need to: 
* Add DI/DO marking
* Add a new color for this part for the MISO/MOSI markings indicating that they are the pinout for ISP programming only. 

This would need to be done by @MCUdude though, he is the one who created these for me.

MCUdude, 2017-02-20:
I'll fix this soon! If I understand this correctly, DI/DO is for the SPI-like interface, while MOSI/MISO is for ISP programming only?

> However, unlike "real" SPI, USI has DI and DO, not MISO and MOSI. Thus, the input and output lines for ISP programming are opposite what would be used when using the USI as an SPI master.

Does this means that PB1 should also have a **DI** marking, and PB2 should have a **DO** marking?

> Add a new color for this part for the MISO/MOSI markings indicating that they are the pinout for ISP programming only.

I don't think changing the color would be the the best solution. The green-ish color indicates that these pins have some kind of serial interface. Maybe I can do like I've done with the new [MightyCore pinouts](https://github.com/MCUdude/MightyCore#pinout), using * ◼︎◆ symbols?

SpenceKonde, 2017-02-22:
Yeah, PB1 has DI and PB2 has DO..

Issue is how to clearly communicate to people that the "MISO" and "MOSI" markings are only for programming, and that when using it as an SPI master, the in and out lines are swapped...

MCUdude, 2017-02-22:
How about this?
[image]

slavkoja, 2017-02-22:
You cannot communicate to people, when they don't understand the hardware or don't read the HW specification, then don't worry about these. But of course, adding DI/DO (colored as serial interface) is good idea.

For my own pinouts i am using different color for "service" pins (eg. RESET here), then perhaps you can color MOSI/MISO/SCLK by the same color as RESET, to group them together, but i don't do this.

SpenceKonde, 2017-02-22:
Well, guess who gets requests for support and "bug reports" from those people who don't read the datasheet? 

Actually, I rather like the idea of "service pins" or ISP programming pins (incl. reset) being a different color - we're gonna need to apply this change to more pinout diagrams, no matter how we do it. I was looking through the datasheets - and it actually turns out that this issue is present on every Tiny that has a USI! 

They *all* have MISO and MOSI on DO and DI respectively: Tiny1634, Tiny x5, Tiny x4, Tiny x61, Tiny x7, Tiny x313

slavkoja, 2017-02-23:
@SpenceKonde sure, that they will ask here, but i want to tell, that don't matter how perfect your pinouts will be, these requests will appears here from time to time.

You are right, that this affects all USI based devices. Some time ago i wrote small python script to generate colored pinouts with SVG output. It is not perfect and it is limited to DIL/DIP packages, but perhaps can help or can act as inspiration, see it at http://git.slavino.sk/svgpinout.git/ (use master, tagged version has broken setup.py). You can tweak colors and groups of colors as you want/need, see some examples at http://slavino.sk/vyvoj/python/farebny-popis-vyvodov.html (in Slovak only)

SpenceKonde, 2017-02-23:
Yeah - I realize some inquiries like that are unavoidable, but I hope to minimize it - both to save my users time, and to save myself support time. 

That's a pretty cool pinout generator, the results look great - I may eventually want to steal it for something - but I'd need to sort out a way to extend it so I could generate a pinout for a TQFP package; I want to have all the diagrams have a similar visual style.

slavkoja, 2017-02-26:
Perhaps i will sometime add support for pins on four sides, but not in near time, i am sorry.

MCUdude, 2017-02-26:
> Actually, I rather like the idea of "service pins" or ISP programming pins (incl. reset) being a different color - we're gonna need to apply this change to more pinout diagrams, no matter how we do it.

Any thoughts on how we should solve this? The SCK pin is used for both USI and ISP programming. What color should this have?

MCUdude, 2017-02-26:
What do you think about this? Should be a quick job to apply this to the other USI enabled devices
[image]

SpenceKonde, 2017-02-27:
That looks great! I think that's the way to go.

MCUdude, 2017-02-27:
Great! I'll add this to the other USI enabled devices, and upload a ZIP file soon 👍

MCUdude, 2017-03-03:
Here's a ZIP file with all pinouts. Give me a sign if you find any errors

[ATTiny_pinout.zip](https://github.com/SpenceKonde/ATTinyCore/files/817097/ATTiny_pinout.zip)

SpenceKonde, 2017-03-08:
Thanks - those look great. My life has been kinda crazy lately so I haven't had a chance to process them and put them up. Will do so ASAP - worst case, I'll get it done on saturday.

SpenceKonde, 2017-03-14:
Images updated. Currently updating all the references,  since both github and tindie cache images, so I have to upload them under a different name to make them update.

commit fa4385d2 (2017-03-14): New pinout diagrams (fix #110)
closed 2017-03-14 by SpenceKonde