SpenceKonde/ATTinyCore #111: tone() doesn't give the expected output if repeatedly called with the same argument in a tight loop

SpenceKonde, 2017-02-15:
Technically, one can argue that this is bad coding to begin with, to spam calls to tone() as fast as the processor can do it - but people do it all the time, so we should handle it. 

Full story is here
http://forum.arduino.cc/index.php?topic=447949

sleemanj, 2017-02-15:
FWIW, I rewrite tone for the x5 as variant code...

https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tinyX5/pins_arduino.h#L72
https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tinyX5/Tone.cpp

I also implemented tone() for Tiny13, but the implementation is a bit more complex to fit within the flash and ram of that chip by doing more compile-time calculation

https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tiny13/pins_arduino.h#L75
https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tiny13/Tone.h
https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tiny13/Tone.cpp

SpenceKonde, 2017-02-15:
That implementation doesn't support generating output directly with output compare though - which I think is pretty cool, and was where a fair amount of my effort went. You can generate an absurdly high frequency square wave if you clock timer1 off the PLL

sleemanj, 2017-02-15:
Yeah, if that's what you need then yep, I'm only targeting the lowest common denominator - like I say in my comments "let's face it, blurting out Mario Bros is about the most people use tone() for"

But with that said, I would encourage putting `tone()` in variant code specially crafted for the, err, variant, if you are going to be digging in it anyway and ripping out the `#if` mess that doesn't apply to that variant.  

Working out exactly which bit of code applies to your specific situation when it's a sea of `#if` statements based around the existence or otherwise of certain registers... well you know, not fun.

DaleMitchell, 2017-03-09:
I was about to open an issue but then I saw this one... is this screenshot the behavior this issue is discussing?

The glitch I have encountered is independent of the given frequency (happens all the way between 15 kHz and 65 kHz).

### 16.7 kHz
[image]

SpenceKonde, 2017-03-09:
Not sure if yours is a manifestation of the same issue or not, as described in the link at the top of this issue, the problem reported here was that someone put the tone() call in a busy-wait loop and got essentially nothing out. 

 What microcontroller are you using? There are several almost completely different codepaths for tone() depending on the microcontroller you're using? Can you post code that reproduces the issue? If it's a tiny85, which pin are you using? Are you using pin4 as recommended? 

Do you not see this behavior with tone() on an atmega328p-based board? I have not played with tone() at all (I don't even own a peizo buzzer) so I don't really know what the baseline expectation is there. This is something that's going to take some digging to sort out I think, but it sounds like you're encountering a much more serious issue than what initiated this issue (I was not planning to prioritize this - continual calls to tone() within a super-tight while loop is a corner case, not worthy of development effort that could be applied to the unified wire library (#53) or the enhanced fuse configuration options (#105), or the bootloader hex file export issue (#73). But it sounds like there may be a much larger and more important issue with tone here)

I would greatly appreciate your answers to these questions so that I can make the best use of any scraps of time I'm able to find to take a look at this.

SpenceKonde, 2017-03-17:
@DaleMitchell - any update? I may have a chance to look at this over the weekend, and it would greatly improve my chances of getting this fixed if you could answer some of the questions I posed above. Thanks!

DaleMitchell, 2017-03-18:
Sorry for the delay, it's been a busy weekend + week. I hope you had a good Saint Patrick's Day!

> What microcontroller are you using?

I'm using the **ATTiny44**. The glitch does not occur on the Arduino Uno with the same code.  

> Can you post code that reproduces the issue? Are you using pin4 as recommended?

(without the "if analog reading has changed" mitigation):
```int inputPin = A0;
unsigned int reading = 0;
unsigned int frequency = 0; 
int outputPin = 4;

void setup() {

}

void loop() {
  reading = analogRead(inputPin);
  frequency = map(reading, 0, 1023, 16000, 65000);
  tone(outputPin, frequency);
  delay(1);
}
```

commit fbe389b8 (2018-10-09): fix #111

We now disregard calls to tone() if no duration is specified, and the pin and frequency are unchanged from the previous call. This prevents glitches when tone() is called with the same parameters as the currently playing tone.
closed 2018-10-09 by SpenceKonde