SpenceKonde/ATTinyCore #112: Getting some compile errors using SPI

JPG-Consulting, 2017-02-27:
IDE: 1.8.1
OS: Windows 10

Errors:

`ATTinyCore\hardware\avr\1.1.2\libraries\SPI/SPI.h:146:16: error: 'SPE' was not declared in this scope` 

`ATTinyCore\hardware\avr\1.1.2\libraries\SPI/SPI.h:146:27: error: 'MSTR' was not declared in this scope`

`ATTinyCore\hardware\avr\1.1.2\libraries\SPI/SPI.h:146:65: error: 'DORD' was not declared in this scope`

`ATTinyCore\hardware\avr\1.1.2\libraries\SPI/SPI.h:203:5: error: 'SPCR' was not declared in this scope`

`ATTinyCore\hardware\avr\1.1.2\libraries\SPI/SPI.h:204:5: error: 'SPSR' was not declared in this scope`

`ATTinyCore\hardware\avr\1.1.2\libraries\SPI/SPI.h:209:5: error: 'SPDR' was not declared in this scope`

`ATTinyCore\hardware\avr\1.1.2\libraries\SPI/SPI.h:217:25: error: 'SPIF' was not declared in this scope`

SpenceKonde, 2017-02-27:
That's because you're trying to use the SPI library on a chip that does not have an SPI peripheral. Most of the Tinies don't have SPI, just a USI. 

There is an ongoing project to create "universal" versions of SPI and Wire described here: 
https://github.com/SpenceKonde/ATTinyCore/issues/53

As described in the comments over there, we may have a working version of SPI in the Unified-SPI branch (you need to uninstall the board manager version and install from github) - I know it compiles, but I haven'd had time to test whether it actually works on devices with a USI.  If you try it out, please let me know how you get on with it - I think it works, and if it does, the only thing stopping it from being merged in is that nobody has tested it out and vouched for it working.

SpenceKonde, 2017-03-01:
Closing as duplicate of #53 - if you try that out, please let me know in the comments over there how you get on with it.

closed 2017-03-01 by SpenceKonde