SpenceKonde/ATTinyCore #113: ParseINT not available in 1.1.2 ATTINY85

kutysam, 2017-03-18:
> #include <SoftwareSerial.h>
> SoftwareSerial mySerial(0, 1); // RX, TX
> void setup(){
>   mySerial.begin(9600);
>   mySerial.parseInt();
> }
> void loop(){}

Running the above code gives me this error

> Using library SoftwareSerial at version 1.0 in folder: C:\Users\xxx\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.2\libraries\SoftwareSerial 
> exit status 1
> 'class SoftwareSerial' has no member named 'parseInt'SoftwareSerialExample:5:` error: 'class SoftwareSerial' has no member named 'parseInt'

However, using "https://github.com/damellis/attiny" core it works fine as it uses the default softwareserial library
> Using library SoftwareSerial at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial

sleemanj, 2017-03-18:
These methods are part of [Stream](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Stream.h) ( [Implementation](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Stream.cpp) ), for whatever reason they have never been brought into TinyCore where Stream only has some virtual functions.

Probably these could be just brought straight across, but those two unsigned longs would be a wasted pain on small chips if you didn't use anything needing them, but maybe the optimizer would be smart enough to not allocate them?

kutysam, 2017-03-19:
I have a valid use case for it, as per others whom use this parseint function.

Basically I'm reading from a serial interface 12,55,33,88
And I plan to put them into an array . Thus the easiest way would be to parseint as it stops at ,

Though long is used , it is also being reused! So memory isn't really 'wasted'

If we don't have this function, it's going to be painful as I have to 356,789
Read by char and.. find the number of digits.. and then 3x100+5x10+6x1.

I certainly do hope this function gets implemented ! Thank you!

SpenceKonde, 2017-03-19:
Try using the github version; I think this was fixed with https://github.com/SpenceKonde/ATTinyCore/commit/64ceae892d2dfb38a510cb7257dbcb87904cf28a

sleemanj, 2017-03-19:
@SpenceKonde yes looks like you added them then.  

@kutysam 

With some of the ATTiny chips SRAM is very limited, even on a tiny25 these extra 8 bytes (two 4 byes longs) is 6% of the available memory (and on a Tiny10 it's 25% of it's 32 bytes available SRAM).  Something to be aware of.

closed 2017-03-20 by SpenceKonde