SpenceKonde/ATTinyCore #114: Serial output gets messy using internal 8MHz

JPG-Consulting, 2017-03-26:
I'm running the following code:

    const int tempPin = PA0;
    int value;
    
    void setup() {
      // put your setup code here, to run once:
      Serial.begin(9600);
      analogReference(INTERNAL);
      Serial.println("Ready");
    }
    
    void loop() {
      int reading = 0;
    
      reading = analogRead(tempPin); 
      Serial.print("Reading: ");
      Serial.println(reading);
      delay(20);
    }

There is an LM35 attached to A0 8i understand this is phisical pin 13, just under GND). My problems:

- If set to 8MHz Internal oscillator the Serial output gets unreadable. At 1MHz it shows ok.

- analogRead(tempPin) always returns 0. using the multimeter the LM35 is outputing 0.19, o another test it was outputing 0.21... however analogRead keeps saying the value is 0.

closed 2017-03-26 by JPG-Consulting

nedoskiv, 2019-03-08:
That saved the day, I got no output at all, switching to 1mhz and I can read out, thank you

SpenceKonde, 2019-03-08:
My guess in both of these cases is that you are setting the tools menu clock speed to 8MHz internal, but have forgotten to do "burn bootloader" to set the fuses so that the chip actually runs at 8MHz. On boards without a bootloader, this sets fuses only, on boards with a bootloader, it sets fuses and writes bootloader to flash (this is the same behavior as in the official arduino avr board package and other major third party hardware packages - we never set fuses unless you choose "burn bootloader", which I recognize can be confusing when working with a chip that has no bootloader - I didn't set this convention). 

So the chip is running code compiled for 8MHz at 1MHz (the default for a virgin chip), and all timing (including baud rates) is off by a factor of 8. Once you select the desired clock settings, connect the ISP programmer, and choose "burn bootloader" before uploading your sketch, and it should work at 8MHz (be sure to choose internal 8MHz unless using an external crystal - choosing external crystal clock source without one connected will make it difficult to recover the chip)

nedoskiv, 2019-03-09:
I'm new in programming attiny, never burn bootloader to them When I choose that option from where it take the bootloader code to burn it?

SpenceKonde, 2019-03-09:
It is supplied with the core. If a (no bootloader) option is selected, the code uploaded to the flash is a dummy file (this is due to the limitations of the IDE). If an (optiboot) option is selected, it uses the included optiboot hex file (exact file depends on other selected options). 

However, as I said above, the ONLY TIME the fuses are written is when you do "burn bootloader" - and the fuses determine which clock source will be used. So you must do "burn bootloader" to set the fuses to run at 8MHz Internal. Then you can upload normally with the 8MHz option selected.

nedoskiv, 2019-03-09:
changed clock to 8mhz and burn bootloader, serial works fine, even i2c interface start working (now working when was at 1mhz)
thank you