SpenceKonde/ATTinyCore #116: Attiny84 pinMapping issue

battosai30, 2017-04-08:
Hi,

I just observed an issue with Attiny84 (so I guess on 24/44 too) :

http://drazzy.com/e/img/PinoutT84a.jpg

When I select "Damellis core" pin mapping, I get clockwise pin mapping in reality (I did not check if the inverse is true).

per1234, 2017-04-08:
You didn't respond to the question I asked you on the other issue report(https://github.com/arduino/Arduino/issues/6160):
>It's not exactly clear to me what the issue you're reporting is. Are you saying that you think the label on the pinout diagram "Alternative pinout" for the "Counterclockwise (like ATTinyCore)" pin mapping is confusing as you would expect the "Clockwise (like damellis core)" pin mapping to be considered the "alternative"? It was [specifically decided](https://github.com/SpenceKonde/ATTinyCore/issues/50#issuecomment-202126223) that damellis mapping would be the "default", though that's also confusing since the ATTinyCore mapping is actually the default selection in the **Tools > Pin Mapping**.  It does seem like it would be more clear for the label on the diagram to match the text of the options in the **Tools > Pin Mapping** menu rather than using the vague term "Alternative pinout".

battosai30, 2017-04-08:
When I explicitly select AttinyCore pinMapping, I do not get it "in real" (I did not check at this moment if at this moment I got the damellisCore pinMapping). To get my circuit working, I had to select damellisCore pin mapping in arduino IDE, and use in my Arduino code the pins numbers of AttinyCore pin mapping.

SpenceKonde, 2017-04-08:
I haven't touched this in eons. Was this always broken?!

battosai30, 2017-04-08:
I think at least since this summer (I made another project based on attiny84 and I noticed this bug but I was very busy so did not noticed this issue on git)

SpenceKonde, 2017-04-24:
I think I fixed the default selection issue. 

However, I'm just staring at the files, and I can't understand how you're ending up with the wrong pin mapping... the files sure look correct...

battosai30, 2017-04-24:
Just to precise : my installation is completly new (just formated my HDD).

It's strange (and mybe not linked with this one) but on another project with Attiny441 I meet pin mapping problems too. I will do more tests to confirm it,  but it seems analog pin names and digital pin names are "mixed" : if I call pinMode(A7,INPUT) pin 7 and not A7, and 7 is configured as INPUT.

SpenceKonde, 2017-04-24:
digital pin functions (including pinMode) must use the digital pin number, not the analog pin number. I think I have a warning about this. 

A7 (for example) is just #defined as 7 (the analog channel number), so pinMode() can't tell that you meant A7; all it see's is the 7. This isn't ideal, but changing it (by putting an offset on the Ax defines and removing it in analogRead() ) is tricky, because it needs to be done without breaking the ability to read an analog channel other than an analog pin just by passing the channel number (for example, differential inputs, or reading the reference in order to measure the battery voltage). This should be in a separate issue in any case.

SpenceKonde, 2017-05-13:
Updated issue. 

I cannot reproduce issues described above with the pin mapping, though I did fix an issue with a few of the PIN_xn defines.

SpenceKonde, 2017-05-22:
I really want to get this into 1.1.3 - will take a stab at it tonight. This is causing a lot of confusion.

battosai30, 2017-05-23:
Great news :)

SpenceKonde, 2017-06-05:
This may fix it for x4, x5, x313, x61, x7, x8 (not x41, 1634 or 828) https://github.com/SpenceKonde/ATTinyCore/tree/FixDigitalAnalogPinNumberIssues

SpenceKonde, 2017-07-06:
Fix for non x41/1634/828 versions tested, other versions written but untested.

SpenceKonde, 2017-07-10:
Fix was tested last night, merged, and will be in 1.1.3

closed 2017-07-10 by SpenceKonde

sleemanj, 2017-07-24:
@AriUJ raises a rather good point here...

https://github.com/SpenceKonde/ATTinyCore/commit/778099585d28031b4acfe2598a04c6afeeda5fd6#commitcomment-23258894

as far as I can see (have not actually tested), the current release has all the Ax constants set to 0, so analogRead(Ax) isn't going to work for anything other than channel zero, and digitalRead/Write/pinMode for any Ax will only affect pin 0.