SpenceKonde/ATTinyCore #123: Request: RadioHead library support

Thijxx, 2017-04-24:
The [arduino-tiny](https://code.google.com/archive/p/arduino-tiny/) core supports it but for the rest is not as good as this core. It has no SoftwareSerial for example. It also looks abandoned.

More about the library [here](http://www.airspayce.com/mikem/arduino/RadioHead/classRH__ASK.html).

Because many projects use 433MHz and the ATtiny85, this feature request may be worth the trouble 😀

SpenceKonde, 2017-04-24:
Yeah, arduino-tiny hasn't been maintained in eons; If I understand the family tree correctly, this core is based off of arduino-tiny (several maintainers ago) - this core was made from the ashes of three unmaintained attiny cores that between them had support for most of the tiny's. 

I would expect RadioHead to work without issue on this - nothing in RadioHead should work differently on my core vs other cores, so if there's any difference, it's probably something simple and silly. Which is good, because the code underlying RadioHead is too deep for me!

Any assistance you can give here is valuable; You've got a testbed set up already, while I don't. If I have to start from scratch, I won't be able to get to it probably until june (I'm flat out with two client projects and a major event next month)... but if you can get a PR, or even just a description of what needs to be fixed, I could probably get the changes in sooner.

Thijxx, 2017-04-24:
Thanks, I will describe the problem as good as I can and I can test or try things if needed.
I've included the sketch and the log or the error. A lot of hardware SPI errors.

What I can do is compare your core with the arduino-tiny core to see where it differs, it would help if you give me some direction :)

**The sketch**
```
#include <RH_ASK.h>

void setup() {
}

void loop() {
}
```

**The error**
```
Compiling sketch...
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/RadioHead" "/tmp/arduino_build_608897/sketch/sketch_apr24a.ino.cpp" -o "/tmp/arduino_build_608897/sketch/sketch_apr24a.ino.cpp.o"
In file included from /home/thijs/Documents/Arduino/libraries/RadioHead/RHGenericDriver.h:9:0,
                 from /home/thijs/Documents/Arduino/libraries/RadioHead/RH_ASK.h:9,
                 from /tmp/arduino_modified_sketch_587021/sketch_apr24a.ino:1:
/home/thijs/Documents/Arduino/libraries/RadioHead/RadioHead.h:786:4: warning: #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead. [-Wcpp]
   #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead.
    ^
Compiling libraries...
Compiling library "RadioHead"
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/RadioHead" "/home/thijs/Documents/Arduino/libraries/RadioHead/RHCRC.cpp" -o "/tmp/arduino_build_608897/libraries/RadioHead/RHCRC.cpp.o"
In file included from /home/thijs/Documents/Arduino/libraries/RadioHead/RHCRC.h:12:0,
                 from /home/thijs/Documents/Arduino/libraries/RadioHead/RHCRC.cpp:43:
/home/thijs/Documents/Arduino/libraries/RadioHead/RadioHead.h:786:4: warning: #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead. [-Wcpp]
   #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead.
    ^
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/RadioHead" "/home/thijs/Documents/Arduino/libraries/RadioHead/RHDatagram.cpp" -o "/tmp/arduino_build_608897/libraries/RadioHead/RHDatagram.cpp.o"
In file included from /home/thijs/Documents/Arduino/libraries/RadioHead/RHGenericDriver.h:9:0,
                 from /home/thijs/Documents/Arduino/libraries/RadioHead/RHDatagram.h:9,
                 from /home/thijs/Documents/Arduino/libraries/RadioHead/RHDatagram.cpp:6:
/home/thijs/Documents/Arduino/libraries/RadioHead/RadioHead.h:786:4: warning: #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead. [-Wcpp]
   #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead.
    ^
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/libraries/RadioHead" "/home/thijs/Documents/Arduino/libraries/RadioHead/RHGenericDriver.cpp" -o "/tmp/arduino_build_608897/libraries/RadioHead/RHGenericDriver.cpp.o"
In file included from /home/thijs/Documents/Arduino/libraries/RadioHead/RHGenericDriver.h:9:0,
                 from /home/thijs/Documents/Arduino/libraries/RadioHead/RHGenericDriver.cpp:6:
/home/thijs/Documents/Arduino/libraries/RadioHead/RadioHead.h:786:4: warning: #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead. [-Wcpp]
   #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead.
    ^
"/home/thijs/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny" "-I/home/thijs/Documents/Arduino/hardware/ATTinyCore-master/avr/variants/tinyX5" "-I/home/thijs/Documents/Arduino/ [... cut ...]
/home/thijs/Documents/Arduino/libraries/RadioHead/RH_Serial.cpp:233:12: error: invalid use of incomplete type 'class HardwareSerial'
     _serial.write(ch);
            ^
In file included from /home/thijs/Documents/Arduino/libraries/RadioHead/RH_Serial.cpp:6:0:
/home/thijs/Documents/Arduino/libraries/RadioHead/RH_Serial.h:40:7: error: forward declaration of 'class HardwareSerial'
 class HardwareSerial;
       ^
Using library RadioHead in folder: /home/thijs/Documents/Arduino/libraries/RadioHead (legacy)
exit status 1
Error compiling for board ATtiny25/45/85.

```

SpenceKonde, 2017-04-24:
Where does RadioHead make the decision to include RH_Serial.h? I can't immediately find it in that ineffable #ifdef hell...

Thijxx, 2017-04-25:
There is never any include for `RH_Serial.h` in any file in the library. I do not understand how this gets in. The only link between **RH_ASK** that I include and **RH_Serial** is this line in `RH_serial.h` that names a class that is also defined in **RHGenericDriver** which is in turn included in **RH_ASK**.

`class RH_Serial : public RHGenericDriver`

**includes**
```
RH_ASK
-> RHCRC
-> RHGenericDriver
--> RadioHead
```

SpenceKonde, 2017-04-25:
Honestly, I would suggest this for 8-bit AVR devices - because Atmel names the registers consistently across the 8-bit AVR product line, we can just check whether the data register for SPI or USART is defined:

* To check if a chip has hardware SPI, check if SPDR is defined. 
* To check if a chip has hardware UART, check if UDR or UDR0 is defined

That way, you don't have to put defines in for every part. 

I'll bet the problem with RH_Serial is related to the fact that there does exist a HardwareSerial.h in the core - to support the chips that do have it.

Thijxx, 2017-04-25:
Since the issue is clear and I let the RadioHead poeple know what could be a fix I will close this issue. Also, if anyone reads this looking for RadioHead ASK on ATtiny, check out [TinyHead](https://github.com/Thijxx/TinyHead), made a RadioHead ASK 'light' edition for the Tinies

closed 2017-04-25 by Thijxx