SpenceKonde/ATTinyCore #124: Platformio Compatibility

chappy1978, 2017-04-25:
Its seems there are a few bugs when programming ATTiny's using Platformio and the standard atmelavr packages. (Example: warning: "BIN" redefined #define BIN 2) I know its just a warning but who wants to look at that every time they build.  I thought it would be nice to use ATTinyCore. After looking through Platformio's folders the folder format is very different, as compared to Arduino. How would I go through creating a package install for Platformio?

SpenceKonde, 2017-04-25:
I am not familiar with platformio... do we have any platformio users in the house who can speak to the feasibility of this?

chappy1978, 2017-04-25:
It a pretty good IDE. Open-Source and well maintained. The difficult part is it uses a system JSONs not a GUI board manager like Arduino.

Thijxx, 2017-04-26:
I've used platformio extensively for Arduino and ATtiny projects.  Half of the libraries I use do not run using platformio and run fine using the Arduino IDE, the IDE is great but it created many new nightmares. So, on topic: I'd like to help getting the core available on platformio! On the forum I found this `place your core in  ~/.platformio/packages/framework-arduinoavr/cores`. I wonder how we can test if the core is the right one?

chappy1978, 2017-04-26:
Morning,
  I got ATTinyCore to work with Platformio all be it manually. Also wrote myself some notes on it. I can send that to you. I check the link out when I get a chance. 
  Jeff

chappy1978, 2017-04-26:
> I've used platformio extensively for Arduino and ATtiny projects. Half of the libraries I use do not run using platformio and run fine using the Arduino IDE, the IDE is great but it created many new [...]
Hey,
 So that in indeed where I went to make ATTinyCore work. Below are the basic instructions for the t85. Send me an email, if there is any problems. 

Instructions
How to add AttinyCore to Platformio

First download the master from https://github.com/SpenceKonde/ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>
Second Copy
~/avr/cores/tiny to ~/.platformio/packages/framework-arduinoavr/cores
~/avr/library/ (everything) to ~/.platformio/packages/framework-arduinoavr/libraries/__cores__/tiny
~/avr/variants (everything) to ~/.platformio/packages/framework-arduinoavr/variants
 
Then add the text below to boards.txt
##############################################################

attinyx5.name=ATtiny25/45/85
attinyx5.upload.maximum_size=8192
attinyx5.upload.maximum_data_size=512
attinyx5.upload.tool=avrdude

attinyx5.bootloader.tool=avrdude
attinyx5.bootloader.unlock_bits=0xFF
attinyx5.bootloader.lock_bits=0xFF
attinyx5.bootloader.extended_fuses=0xFF

attinyx5.build.core=tiny
attinyx5.build.board=AVR_ATTINYX5
attinyx5.build.variant=tinyX5

Then go to ~/.platformio/platforms/atmelavr/boards
open attiny85.json
duplicate it and name attiny85x.json
Cut and paste these line to attiny85x.json
{
  "build": {
    "core": "tiny",
    "extra_flags": "-DARDUINO_ARCH_AVR -DARDUINO_AVR_ATTINY85",
    "f_cpu": "8000000L",
    "mcu": "attiny85",
    "variant": "tinyX5"
  },

Now in the platformio.ini of your program change 

board=attiny85
to
board=attiny85x

SpenceKonde, 2017-05-13:
Sounds like there's nothing I can do here, it's all platformio stuff? 

Closing - reopen if my understanding is incorrect.

closed 2017-05-13 by SpenceKonde

Thijxx, 2017-05-15:
platformio updated the attiny core, not sure if it's exactly the setup as your core but it works fine with the libraries I use and with softwareserial etc.