SpenceKonde/ATTinyCore #126: Fix mysterious ISP programming issues

SpenceKonde, 2017-04-28:
In-house development work with USBAsp programmers has been encountering wacky behavior when using "upload using programmer", and I've been forced to fall back to Arduino as ISP to complete urgent projects for my clients. 

It is critical that these problems are understood and remedied before the next release!

bperrybap, 2017-05-12:
You may be having issues with USBasp f/w.
You never know what you are getting in these cheap USBasp devices.
Some devices are using REALLY old f/w, some have modified f/w, and some are using old f/w because the official updated f/w is larger would not fit in the part you are using.
But work on the official f/w has basically stopped. While I've tried to get a hold of fischl, I got no response from emails.
A while back, Two of us did a massive amount of work to generate a new updated version of the code.
We fixed many bugs in the part, and in particular correct how the clocking was being done.
With the original code, it was surprising that it worked as well as it did.
I have pre-built images and it the code with updates and fixes can still be loaded in to the smaller flash parts used on some devices.
It fixes many bugs and does really good auto detect of SCK clock speeds.
I never got around to doing a final release but you can get the code here:
https://github.com/bperrybap/usbasp/tree/1.06-alpha
It is very stable and has been through lots of testing on many different parts and clock speeds.
I guess I should revive it,  put it over on the github "USBasp" project that I grabbed and reserved for it and make an official release for it.

The package includes drag and drop installation scripts for Windows, and linux so installation is easy.
Or you can do it using make as the makefile has many capabilities for building and flash burning that can all be done without having to edit the makefile.
I'd recommend that you update at least one of your USBasp test devices and see if your problems go away.

sleemanj, 2017-05-13:
> I guess I should revive it, put it over on the github "USBasp" project that I grabbed and reserved for it and make an official release for it.

Yes please do, I have been thinking about flashing some USBAsp's with your firmware but held off because I wasn't sure what the authoritative last "stable" version from you was.  You should firm up that firmware release.

SpenceKonde, 2017-05-13:
I'm using your 5-18 f/w. 

It's not an f/w issue though - the avrdude invocation was coming out straight-up wrong. Need to do more experimentation.

bperrybap, 2017-05-13:
That alpha6 release the latest code I have. Unfortunately, you might see some other versions out there with same versioning information but it is different. I need to get with Peter on this - I think I noticed that he did some poking around in his repo lately.
 My preference would be to get him to donate his current to USBasp project and re-clone back to his  own repo for his use/testing. That way the master repo for the project is the orignal/toplevel github commit vs a clone of his repo.
The USBasp repo/project I set up is here: https://github.com/usbasp
The only issue that I know of is that there can be issue for parts running an 8k clock. The autodetected 4k SCK clock rate used by the code doesn't always work due to the timing related to the clocks being slightly off the exact clock rates on either the ISP device or the target. So that is something that I should remove. The only down side is that programming a target part with a clock that slow will take twice as long. Still not bad considering there really isn't a reliable alternative. You can always tell avrdude to use the faster to set it manually if it were a production environment and you knew that the faster clock works.

bperrybap, 2017-05-13:
> It's not an f/w issue though - the avrdude invocation was coming out straight-up wrong. Need to do more experimentation.

From the Arduino IDE? if so which version of IDE? I have every IDE since 0017 on my machine.
I've not experience that, but I do tend to use bootloaders more often and I'm using linux if that matters.

SpenceKonde, 2017-05-13:
AVRdude command lines were in fact correct. I misunderstood the usage of one of the flags passed to avrdude. I assembled a board with just the t441 and necessary supporting components, and problem does not reproduce there, but does reproduce on the original hardware; suspect issue is related to external components loading the pins. But it is not an issue with the core.

closed 2017-05-13 by SpenceKonde