SpenceKonde/ATTinyCore #128: Enabling LTO causes compiler error in Arduino IDE 1.8.2

j54n1n, 2017-05-11:
If the LTO option is enabled I get a compiler error where it says that "-fno-fat-lto-objects are supported only with linker plugin".

This error can be reproduced on my machine with the BareMinimum sketch and with all of the ATtiny boards.

Bellow you can find entire error message:
```
Arduino: 1.8.2 (Windows 7), Board: "ATtiny441/841 (No bootloader), ATtiny841, Enabled, 8 MHz (internal), B.O.D. Disabled"

C:\SDK\arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\SDK\arduino\hardware -hardware C:\Users\j54n1n\AppData\Local\Arduino15\packages -hardware C:\Files\Work\arduino\sketchbook\hardware -tools C:\SDK\arduino\tools-builder -tools C:\SDK\arduino\hardware\tools\avr -tools C:\Users\j54n1n\AppData\Local\Arduino15\packages -built-in-libraries C:\SDK\arduino\libraries -libraries C:\Files\Work\arduino\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx41:chip=841,LTO=enable,clock=8internal,bod=disable -ide-version=10802 -build-path C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183 -warnings=none -build-cache C:\Users\j54n1n\AppData\Local\Temp\arduino_cache_989735 -prefs=build.warn_data_percentage=75 -verbose C:\SDK\arduino\examples\01.Basics\BareMinimum\BareMinimum.ino
C:\SDK\arduino\arduino-builder -compile -logger=machine -hardware C:\SDK\arduino\hardware -hardware C:\Users\j54n1n\AppData\Local\Arduino15\packages -hardware C:\Files\Work\arduino\sketchbook\hardware -tools C:\SDK\arduino\tools-builder -tools C:\SDK\arduino\hardware\tools\avr -tools C:\Users\j54n1n\AppData\Local\Arduino15\packages -built-in-libraries C:\SDK\arduino\libraries -libraries C:\Files\Work\arduino\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx41:chip=841,LTO=enable,clock=8internal,bod=disable -ide-version=10802 -build-path C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183 -warnings=none -build-cache C:\Users\j54n1n\AppData\Local\Temp\arduino_cache_989735 -prefs=build.warn_data_percentage=75 -verbose C:\SDK\arduino\examples\01.Basics\BareMinimum\BareMinimum.ino
Using board 'attinyx41' from platform in folder: C:\Files\Work\arduino\sketchbook\hardware\ATTinyCore\avr
Using core 'tinymodern' from platform in folder: C:\Files\Work\arduino\sketchbook\hardware\ATTinyCore\avr
Build options changed, rebuilding all
Detecting libraries used...
"C:\Users\j54n1n\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX41 -DARDUINO_ARCH_AVR   "-IC:\Files\Work\arduino\sketchbook\hardware\ATTinyCore\avr\cores\tinymodern" "C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183\sketch\BareMinimum.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\j54n1n\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX41 -DARDUINO_ARCH_AVR   "-IC:\Files\Work\arduino\sketchbook\hardware\ATTinyCore\avr\cores\tinymodern" "C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183\sketch\BareMinimum.ino.cpp" -o "C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\SDK\arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\j54n1n\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX41 -DARDUINO_ARCH_AVR   "-IC:\Files\Work\arduino\sketchbook\hardware\ATTinyCore\avr\cores\tinymodern" "C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183\sketch\BareMinimum.ino.cpp" -o "C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183\sketch\BareMinimum.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\Users\j54n1n\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=attiny841 -DF_CPU=8000000L -DARDUINO=10802 -DARDUINO_AVR_ATTINYX41 -DARDUINO_ARCH_AVR   "-IC:\Files\Work\arduino\sketchbook\hardware\ATTinyCore\avr\cores\tinymodern" "C:\Files\Work\arduino\sketchbook\hardware\ATTinyCore\avr\cores\tinymodern\WInterrupts.c" -o "C:\Users\j54n1n\AppData\Local\Temp\arduino_build_52183\core\WInterrupts.c.o"
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.

exit status 1
Error compiling for board ATtiny441/841 (No bootloader).
```

SpenceKonde, 2017-05-11:
You've either previously installed an old version of the Arduino AVR board package using board manager prior to installing 1.8.2 and not updated the Arduino AVR board package since upgrading to 1.8.2, or you have used board manager to downgrade the Arduino AVR board package. This core uses the compiler from the official Arduino AVR boards package (there are a number of good reasons for this - this is the typical solution used for the compiler on third party AVR board packages like this).

Thus, the AVR boards package must be at least 1.6.14 (which is what came with 1.6.12 of the IDE), which is when they updated it from 4.8.1 (which doesn't have LTO) to 4.9.2 (which does). So all you need to do is update the official AVR board package through board manager (I recommend to the latest version, which would be the one paired with the 1.8.2 release), and LTO should work normally.

closed 2017-05-11 by SpenceKonde

j54n1n, 2017-05-11:
You are right. There was installed an older version of the compiler tools from the [Digispark AVR Core](https://github.com/digistump/DigistumpArduino/tree/1.6.7/digistump-avr) with its corresponding [package file](https://github.com/digistump/arduino-boards-index/blob/f914873fb56062b4fcf70ac43fcd973db0323a1d/package_digistump_index.json).

At the moment a quick fix is to uninstall the Digispark AVR Boards from within the Board Manager.

P.S.:
I had installed also Watterott AVR Boards but for some reason that package had chosen the right compiler tools (=Arduino built-in tools) with its "Standard Arduino" Core menu setting.
Also the built-in Arduino boards select the original Arduino compiler tools and not the Digispark tools.

So maybe it’s only a question of modifying the settings in the platform or package files!?

j54n1n, 2017-05-19:
See also: https://github.com/arduino/Arduino/issues/5168