SpenceKonde/ATTinyCore #129: link error: undefined reference to `__cxa_pure_virtual'

bperrybap, 2017-05-13:
When I try to use the ATTinyCore package to build my hd44780 library for the attiny85  I get this message from the linker:
```
/tmp/arduino_build_412052/libraries/hd44780/hd44780.cpp.o:(.rodata._ZTV7hd44780[vtable for hd44780]+0xc): undefined reference to `__cxa_pure_virtual'
```
I don’t have this issue with either the mellis core: https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
or the Adafruit core: https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

The difference being the actual core being used by the board entry.
The other board packages use the arduino core vs providing a completely separate core.
I noticed that if I change the core in the boards.txt for the ATtiny25/45/85 board
to attinyx5.build.core=arduino:arduino the link issue goes away.
There is something missing in the tiny core version of the code.
I get the desire for the board package to be completely standalone but Is there any reason why the arduino core can't be used like what the other board packages are doing?

To recreate on ATTinyCore:
Install my hd44780 library from the IDE library manager
Install USIWire from the library manager.
Go down into the ATTinyCore packaging area:
on linux: ~/.arduino15/packages/ATTinyCore/hardware/avr/1.1.2/libraries
Rename the Wire directory to Wire-Orig
Create a new Wire directory with a single Wire.h header file in it with
```
#include <USIWire.h>
```
Create an empty file called Wire.cpp
Make sure to restart the IDE

Try to build the HelloWorld example for the hd44780_I2Cexp i/o class:
[File]->Examples->hd44780->ioClass->hd44780_I2Cexp->HelloWorld
It fails to link.

----------------------------------------------------------------------------------------------------------------------------

To see it work on the Mellis board package:
install the mellis board package
install the USIWire library from the library manager
Go down into the attiny packaging area:
on linux: ~/.arduino15/packages/attiny/hardware/avr/1.0.2/libraries
create a Wire directory 
create a Wire.h header file in it with:
```
#include <USIWire.h>
```
Create an empty file called Wire.cpp
Restart the IDE
select the attiny25/45/85 board type
select the attiny85 processor
select the 8mhz internal clock

Try to build the HelloWorld example for the hd44780_I2Cexp i/o class:
[File]->Examples->hd44780->ioClass->hd44780_I2Cexp->HelloWorld
It works.

chappy1978, 2017-05-13:
Hey
I had a similar problem with another library. Just add 
extern "C" void __cxa_pure_virtual() { while (1); }
To the top of your program and it creates. A loop that is never used. I.e. The problem goes away.

bperrybap, 2017-05-13:
It looks like the official arduino core now includes a file called abi.cpp that declares this and another function called __cxa_deleted_virtual()
They call abort() vs do a hard loop.
I think the ATTinyCore package cores should also provide this for compatibility if it isn't going to share/use the arduino core.
Although, IMO, they should be weak functions, so the user could override them if desired.

bperrybap, 2017-05-13:
@chappy1978 wonking up the sketch to make it work is the wrong solution.
It also won't work for me since this issue is for a library I'm developing (not a sketch) that has to be portable across many platforms where users are writing the sketch code.
The solution is to fix the board package so it works like all the other Arduino board packages and cores.

chappy1978, 2017-05-13:
Slow down there.  It's only a quick patch so you can keep working. Not a solution in itself. I personally don't want to get distracted by every compiler error that pops up. Often times when Im finishing and worked through my original challenges I'll come back and straighten out the small problems.

chappy1978, 2017-05-13:
Here is a good explanation for the problem. 
http://stackoverflow.com/questions/920500/what-is-the-purpose-of-cxa-pure-virtual

bperrybap, 2017-05-13:
I understand and know about pure virtual functions.
I was surprised to see the specific issue show up only with this core.
I wasn't really looking for a patch to keep working as I have working code already. 
The point of this particular issue was to point out the problem to get the board package fixed.

closed 2017-05-13 by SpenceKonde

SpenceKonde, 2017-05-13:
Thanks for reporting this issue. Looks like all it needed was abi.cpp from the official core. 

@bperrybap The official Arduino core is not usable. There are all sorts of issues with it the moment you move off the boring ATTiny's (serial on x7, pinMode on 1634, 841, and 828 come to mind), and I've also added code to support some new menu options setting defines (ex, timer clock source on x5 and x61) - and it's implementation of one of the microsecond timing functions (forget if delayMicroseconds() or micros() is not usable at many clock speeds. 

Fix for this was easy enough; Also, chappy, thanks for the link to explain what the issue is with virtual functions and cxa_pure_virtual; I was not familiar with that issue.

sleemanj, 2017-05-13:
>  I get the desire for the board package to be completely standalone but Is there any reason why the 
> arduino core can't be used like what the other board packages are doing?

The entire raison d'être of ATTinyCore is to provide a core that is designed for supporting a large variety of ATTiny processors to a high degree. 

The Arduino (built in) core is very much **not** designed for Tiny processors and does not support them well (or at all for most), there is some **marginal** support for ATTinyx4 and x5 but that's really about as far as it goes.

The first thing people would discover using arduino:arduino with a tiny that it does "support" is that `Serial` probably doesn't exist or work on chips without hardware UART, like the Tinyx5 - there is no SoftwareSerial in arduino:arduino and certainly no logic to automatically pick SoftwareSerial when the chip doesn't have a Hardware UART - unlike ATTinyCore which ensures that `Serial`  works like people expect it should, by using SoftwareSerial if necessary (and in my fork which supports some even Tinier processors like the Tiny13, 4,5,9 and 10, an even simpler bit-banged half duplex serial).  Plenty of work has also gone into optimising code in ATTinyCore for code-space efficiency, although arguably LTO may now achieve many of the same results, it probably still matters, when your processor has 1kb of flash and a few bytes of ram, space considerations are high, but this does not usually feature in the design decisions of arduino:arduino.

An argument could certainly be made that "arduino:arduino should be worked on to make it support all the processors ATTinyCore does to the level and efficiency that ATTinyCore does, in a clean manner", but there is an infinitely close to zero chance of any such patches to arduino:arduino being accepted by Arduino, also, it would be a big job.

An equal argument could be made that arduino:arduino should be forked and have the tiny support rebased onto it, and indeed that is what happened originally in the dim dark mists of time, unfortunately revision history that far back doesn't exist so it's impractical at least, potentially impossible ( I put some work into trying something like that in the past - https://github.com/sleemanj/Arduino-ATtiny-Core ) to rebase now upon the current arduino:arduino core.

So in short, it is simpler to just add in new features from arduino:arduino, such as this dummy function, as they are needed.

per1234, 2017-05-13:
>there is an infinitely close to zero chance of any such patches to arduino:arduino being accepted by Arduino

I'm not so sure of that. I had previously expected that Arduino would not be interested in their code supporting parts not present on the official boards but when an issue came up in the developer's mailing list a while back regarding changes to some libraries necessary to support ATmega1284P I was surprised to see Massimo Banzi specifically request that pull requests be submitted for those changes. I believe they were all merged without much delay. There is even an official Arduino ATtiny85 board now. It might be worth a feeler PR just to get an idea of how responsive they would be to this.
>also, it would be a big job.

The key would be to submit the pull requests as the changes are made rather than trying to do it later after the history has diverged significantly. I do understand that it's extra work but I think it could pay off in the long term.

These modified cores make me nervous because it takes active involvement to keep them up to date. SpenceKonde does an excellent job of this but if the time ever came when it was not actively maintained an accumulation of issues like this one could cause it to become obsolete. It's not very much work to update hardware definitions when changes to the IDE make that necessary.

SpenceKonde, 2017-05-13:
A big part of why I don't want to use the Arduino core is that I expressly don't want changes to get into the core without being checked out on tinys first.

Consider what would happen without someone like me maintining. Surely it's better to have a "stuck in time" core that works than to have it depend on the Arduino core and break when some change there doesn't work on the tiny's (Which would inevitably happen if someone's not testing - Sometimes I've gotten the impression that the Arduino team doesn't test their code, period, let alone on non-standard parts). If we depended on the Arduino core and it broke something, absent the action of someone to fix it, people would have to download their official core to use ATTinyCore. I think that's a worse outcome.

commit 289bc703 (2017-05-13): fix #129

__cxa_pure_virtual is now provided by abi.cpp as is done in the official
core.

Functions are defined as weak so that they can be overridden per
bperrypap suggestion in #129
commit 0b1e9184 (2017-05-13): Fix #129 for tinymodern too