SpenceKonde/ATTinyCore #138: Maybe a bug? I'm not sure.

srw777, 2017-05-27:
I have a Pololu AVR Programmer (not V2) and have spent the past five hours figuring this out.  I can work around, now, but there's definitely a problem, and I'm not sure where the problem is.  I apologise if it's not part of your core.  I'm definitely a beginner when it comes to using an ISP in Arduino.

Using an Attiny84, I was having no luck at all programming fuses (burn bootloader) or Upload using programmer.  Finally, I loaded Pololu's configuration utility to see if I could find something in there.  I noticed the "ISP Frequency" was set to 750khz, which, according to their documentation, means it's running at 115,200 baud.  Of course, this wouldn't work with the chip running at the factory setting of 1Mhz (internal).  I manually set the frequency to 200khz which corresponds with 57,600.  Burn Bootloader still wouldn't work, and it always resets the "ISP Frequency" to 750khz every time I try.  However, if I set the speed to 200khz, I could use Upload Using Programmer and get a program running at the default 1Mhz.    Finally, I tried the Arduino as ISP and was able to set the fuses and program at 8Mhz internal.  

The reason I dug out the Tiny84 was because I was having problems programming a Tiny841 on a DrAzzy board.  Once I figured out the above, I went back to the Tiny841 board, used the Arduino as ISP to set the fuses to 8Mhz external (I have an 8Mhz crystal on the board) and then I could switch to the Pololu AVR Programmer and everything worked.  (Until I set the fuses back to 1Mhz internal... then I was back to square 1, but now with the knowledge of what to do.)

(In order to use the Pololu AVR Programmer with the ATTinyCore, I had modified the AVRMkII line in programmers.txt to use -P{serial.port} instead of -Pusb.  )

So, I'm not sure what is causing the programmer to always revert to the higher speed.  If it's not the ATTinyCore, could you point me in the right direction?  Or do I simply need a better programmer?

SpenceKonde, 2017-05-31:
It may be that additional parameters can be passed to avrdude to tell it to use a slower SCK speed - but I don't know what those are. I would welcome a pull request to add support for the Pololu AVR Programmer, but as I don't have that hardware to test with, I don't expect to be able to work on this in the near future.

SpenceKonde, 2017-05-31:
Also - I welcome comments from anyone following this repo regarding whether the Pololu AVR Programmer is popular enough to warrant cluttering up the programmers menu further.

srw777, 2017-05-31:
After a few more hours of reading, I realized this is a fairly known thing.  To be honest, don't bother for me.  I've ordered a cheap new programmer.  There's really no point supporting an inferior programmer when I can buy a better one for $2.  Feel free to close it.

closed 2017-05-31 by SpenceKonde