SpenceKonde/ATTinyCore #142: ATtiny85 1mhz Tone()

Hoshwan, 2017-06-05:
Can an attiny85 successfully run Tone() or TimerFreeTone() with a 1mhz clock?

I've actually been using the TimerFreeTone() library with the ATtiny85 which runs beautifully at 8mhz, but not 1mhz.

Different Attiny Cores I've used:

-arduino-tiny

-attiny by David A. Mellis

-ATTinyCore by Spence Kode

I've tried Tone() and TimerfreeTone() with all of the above mentioned cores and none of them work with a 1mhz clock.

Is it possible?

Thanks!

SpenceKonde, 2017-06-05:
Of course Tone() should work. I don't know why it doesn't. Do you have verified test code that is known to reproduce the issue? 

Are you repeatedly calling Tone() in a tight loop? Don't do that.  See #111

The cores other than ATTinyCore and the TimerfreeTone() library are beyond the scope of this repo.

Hoshwan, 2017-06-05:
It works! 
I haven't used Tone() in a while. I just needed to call the function properly. 

1mhz sounds terrible though as compared to 8mhz. (I am rebootloading to switch to the proper fuse setting)
FYI: Using the same core TimerFreeTone() is louder and clearer than Tone() both at 8mhz, but TimerFreeTone() won't work at 1mhz at all.

Any idea why 1mhz Tone() sounds so terrible?

SpenceKonde, 2017-06-05:
Would need to look at what you're seeing on a scope - are you using the recommended output pin? depending on the frequency, you might be pushing the limits of what the processor can due on a pin where it can't use hardware output compare.

Hoshwan, 2017-06-05:
Here's the code:

void setup() {
  tone(1, 2637);
}
void loop()
{
}

here's what the scope looks like:

[image]

Hoshwan, 2017-06-05:
here's the same image without the bottom cut off:
[image]

SpenceKonde, 2017-06-05:
That's strange... 

Do you get better results with pin 4? That should generally work way better for Tone because we can use hardware output compare there instead of interrupts.

Hoshwan, 2017-06-05:
Perfect!
That did it. Pin 4 produces a very clean tone. Too bad the proto PCB is already designed for use with pin 1.  No big deal though, that's what prototypes are for.

I wanted to get it working with a 1mhz clock (off the shelf Attiny85 setting) so the board house doesn't need to set the clock fuse before programming if it's not even necessary.

Digression:
Can you recommend resources for gaining a deeper understanding of AVR architecture and how to apply it? I've been playing with Atmel Studio and exploring using registers directly, but it feels like wading into a very deep ocean without direction or a life jacket.

Any advice you could offer would be appreciated.
Thanks.

SpenceKonde, 2017-06-14:
I gained my understanding of the AVR architecture (actually, stuff like this is more the on-chip peripherals common to AVR chips - not exactly the same thing as the architecture, but this is a matter of semantics) from the part datasheets, for the most part...  I haven't done much of anything with Atmel Studio - I use the IDE, or for a few projects, makefiles. 

The issues with Tone on other pins will be addressed in 1.1.4

SpenceKonde, 2018-10-09:
I couldn't reproduce the original issue. 

However. 1MHz tone did not work as documented. This has been fixed.

commit b1e6812d (2018-10-09): Fix #142, improve tone()

Tone now takes an unsigned long for the frequency, and a few places where this could have overflowed are fixed
closed 2018-10-09 by SpenceKonde