SpenceKonde/ATTinyCore #143: Is it possible to add an option to the Arduino IDE menu for Memory Lock Bits? (i.e., prevent flash from getting dumped over ISP)

DaleMitchell, 2017-06-09:


SpenceKonde, 2017-06-14:
This is tricky, as it would require a more complicated upload recipe - either setting the lockbits every time a sketch is uploaded over ICSP always (either into unlocked or locked mode), OR invoking avrdude again after uploading and verifying.

I consider it a low priority, as the IDE is a development tool unlikely to be used for programming products in production, while the lockbits are rarely used during development. One can always invoke avrdude from the commandline to set the lockbits if needed.

SpenceKonde, 2017-10-20:
I have no particular desire to implement this; if you're going into production, you're not going to be programming units through the IDE anyway, and if you're not in production, you don't need the lockbits. It's a mess to implement for something that is not useful. 

Will document with example avrdude commandlines in FAQ.

closed 2018-01-26 by SpenceKonde