SpenceKonde/ATTinyCore #144: Interrupts not working with ATTiny44

DaleMitchell, 2017-06-13:


SpenceKonde, 2017-06-14:
To expedite fixing this, can you post the test code that reproduces this issue?

gtalusan, 2017-10-15:
I fell into this trap.  I think the confusion stems from Arduino letting you attach an ISR to almost any pin on the rising/falling edge.  On the tiny's, you only get this capability only with PCINT0 as far as I can tell.

For other pins, you need to enable the pin change interrupt like so:
```
GIMSK |= (1 << PCIE);
```
and then enable your pin for change detection:
```
PCMSK |= (1 << PCINT{0,1,2,3,4,5});
```

Patch in your ISR like so:

```
ISR(PCINT0_vect)
{
	int value = digitalRead(pin);
        // figure out state of pin
}
```

SpenceKonde, 2017-10-15:
attachInterrupt does not work for PCINTs. See https://www.arduino.cc/en/Reference/AttachInterrupt - Making it work with PCINTs would interfere with many commonly used libraries (such as software serial, radiohead/virtualwire, etc) because that ISR definition would conflict with that same ISR defined in the the library. 

On the ATTiny x4 series, the only supported pin for attachInterrupt is INT0 (pin 2 or 8 in arduino pin mapping, depending on the pin mapping selected).

Libraries exist for using PCINT interrupts, though I recommend against using them - writing the ISR directly is not significantly harder. PCINT interrupts never support rising/falling - only "change"; it is up to the application to determine which pin was triggered how (often there are ways specific to the application that this can be simplified, which is important for time critical tasks typically used for interrupts.

SpenceKonde, 2017-10-20:
This needs to be tested, documented and either closed, fixed, or recognized as a bug and the details described herein. 

I predict the bug will not reproduce for me, and that the reporting individual was thrown off by some sort of weirdness - maybe the stupid syntax of attachInterrupt(), or maybe something else.

DaleMitchell, 2017-10-20:
> the reporting individual was thrown off by some sort of weirdness

Yup.

closed 2017-10-20 by DaleMitchell

SpenceKonde, 2017-10-20:
Glad to hear you got it sorted out! 

I will add some mention to the FAQ to try to point people in the right direction here.