SpenceKonde/ATTinyCore #145: ATtiny48 not found?!

beicnet, 2017-07-04:
Hi there,

I'm trying to flash an empty sketch to my brand new ATtiny48-PU (have 10pcs of them), and after successfully compiling I got this error message:

```
Sketch uses 334 bytes (8%) of program storage space. Maximum is 4096 bytes.
Global variables use 9 bytes (3%) of dynamic memory, leaving 247 bytes for local variables. Maximum is 256 bytes.

avrdude: AVR Part "attiny48" not found.

Valid parts are:
  uc3a0512 = AT32UC3A0512
  c128     = AT90CAN128
  c32      = AT90CAN32
  c64      = AT90CAN64
  pwm2     = AT90PWM2
  pwm216   = AT90PWM216
  pwm2b    = AT90PWM2B
  pwm3     = AT90PWM3
  pwm316   = AT90PWM316
  pwm3b    = AT90PWM3B
  1200     = AT90S1200
  2313     = AT90S2313
  2333     = AT90S2333
  2343     = AT90S2343
  4414     = AT90S4414
  4433     = AT90S4433
  4434     = AT90S4434
  8515     = AT90S8515
  8535     = AT90S8535
  usb1286  = AT90USB1286
  usb1287  = AT90USB1287
  usb162   = AT90USB162
  usb646   = AT90USB646
  usb647   = AT90USB647
  usb82    = AT90USB82
  m103     = ATmega103
  m128     = ATmega128
  m1280    = ATmega1280
  m1281    = ATmega1281
  m1284    = ATmega1284
  m1284p   = ATmega1284P
  m1284rfr2 = ATmega1284RFR2
  m128rfa1 = ATmega128RFA1
  m128rfr2 = ATmega128RFR2
  m16      = ATmega16
  m161     = ATmega161
  m162     = ATmega162
  m163     = ATmega163
  m164p    = ATmega164P
  m168     = ATmega168
  m168p    = ATmega168P
  m168pb   = ATmega168PB
  m169     = ATmega169
  m16u2    = ATmega16U2
  m2560    = ATmega2560
  m2561    = ATmega2561
  m2564rfr2 = ATmega2564RFR2
  m256rfr2 = ATmega256RFR2
  m32      = ATmega32
  m324p    = ATmega324P
  m324pa   = ATmega324PA
  m325     = ATmega325
  m3250    = ATmega3250
  m328     = ATmega328
  m328p    = ATmega328P
  m329     = ATmega329
  m3290    = ATmega3290
  m3290p   = ATmega3290P
  m329p    = ATmega329P
  m32m1    = ATmega32M1
  m32u2    = ATmega32U2
  m32u4    = ATmega32U4
  m406     = ATMEGA406
  m48      = ATmega48
  m48p     = ATmega48P
  m48pb    = ATmega48PB
  m64      = ATmega64
  m640     = ATmega640
  m644     = ATmega644
  m644p    = ATmega644P
  m644rfr2 = ATmega644RFR2
  m645     = ATmega645
  m6450    = ATmega6450
  m649     = ATmega649
  m6490    = ATmega6490
  m64rfr2  = ATmega64RFR2
  m8       = ATmega8
  m8515    = ATmega8515
  m8535    = ATmega8535
  m88      = ATmega88
  m88p     = ATmega88P
  m88pb    = ATmega88PB
  m8u2     = ATmega8U2
  t10      = ATtiny10
  t11      = ATtiny11
  t12      = ATtiny12
  t13      = ATtiny13
  t15      = ATtiny15
  t1634    = ATtiny1634
  t20      = ATtiny20
  t2313    = ATtiny2313
  t24      = ATtiny24
  t25      = ATtiny25
  t26      = ATtiny26
  t261     = ATtiny261
  t28      = ATtiny28
  t4       = ATtiny4
  t40      = ATtiny40
  t4313    = ATtiny4313
  t43u     = ATtiny43u
  t44      = ATtiny44
  t45      = ATtiny45
  t461     = ATtiny461
  t5       = ATtiny5
  t84      = ATtiny84
  t85      = ATtiny85
  t861     = ATtiny861
  t88      = ATtiny88
  t9       = ATtiny9
  x128a1   = ATxmega128A1
  x128a1d  = ATxmega128A1revD
  x128a1u  = ATxmega128A1U
  x128a3   = ATxmega128A3
  x128a3u  = ATxmega128A3U
  x128a4   = ATxmega128A4
  x128a4u  = ATxmega128A4U
  x128b1   = ATxmega128B1
  x128b3   = ATxmega128B3
  x128c3   = ATxmega128C3
  x128d3   = ATxmega128D3
  x128d4   = ATxmega128D4
  x16a4    = ATxmega16A4
  x16a4u   = ATxmega16A4U
  x16c4    = ATxmega16C4
  x16d4    = ATxmega16D4
  x16e5    = ATxmega16E5
  x192a1   = ATxmega192A1
  x192a3   = ATxmega192A3
  x192a3u  = ATxmega192A3U
  x192c3   = ATxmega192C3
  x192d3   = ATxmega192D3
  x256a1   = ATxmega256A1
  x256a3   = ATxmega256A3
  x256a3b  = ATxmega256A3B
  x256a3bu = ATxmega256A3BU
  x256a3u  = ATxmega256A3U
  x256c3   = ATxmega256C3
  x256d3   = ATxmega256D3
  x32a4    = ATxmega32A4
  x32a4u   = ATxmega32A4U
  x32c4    = ATxmega32C4
  x32d4    = ATxmega32D4
  x32e5    = ATxmega32E5
  x384c3   = ATxmega384C3
  x384d3   = ATxmega384D3
  x64a1    = ATxmega64A1
  x64a1u   = ATxmega64A1U
  x64a3    = ATxmega64A3
  x64a3u   = ATxmega64A3U
  x64a4    = ATxmega64A4
  x64a4u   = ATxmega64A4U
  x64b1    = ATxmega64B1
  x64b3    = ATxmega64B3
  x64c3    = ATxmega64C3
  x64d3    = ATxmega64D3
  x64d4    = ATxmega64D4
  x8e5     = ATxmega8E5
  ucr2     = deprecated, use 'uc3a0512'

An error occurred while uploading the sketch

```

Any help would be appropriated! 

Thank you! ;)

sleemanj, 2017-07-04:
Your avrdude does not support the t48

Ensure you are using an up to date version of the Arduino IDE and that in the Boards Manager you update the "Arduino AVR Boards" package.

beicnet, 2017-07-04:
@sleemanj I'm using Arduino IDE v1.8.2 and ATTinyCore v1.1.2

sleemanj, 2017-07-04:
Tools > Board > Board Manager

What does it say for "Arduino AVR Boards Built-In by Arduino version x.x.xx INSTALLED"

beicnet, 2017-07-04:
v1.6.18

beicnet, 2017-07-04:
Need to go for v1.6.19?

SpenceKonde, 2017-07-04:
Be sure to select a programmer marked with (attiny) per documentation. Due
to a limitation of the IDE, the programmer definitions supplied by the
default core always use the default platform.txt, not the one I supply
which points to the good version of avrdude.conf that I supply instead of
the default version, which doesn't include part definitions for most of the
attiny line.

beicnet, 2017-07-04:
So, what should I choose?

My current config is like on the picture shown:

[image]

SpenceKonde, 2017-07-04:
Arduino/leo as ISP (attiny)

beicnet, 2017-07-04:
Ohhh, I choosed "Arduino/Leo as ISP (ATtiny)" and it's went trough.
Is that a correct "Programmer" choice?

btw... @SpenceKonde there is a typo/free space between y and )... "Arduino/Leo as ISP (ATtiny )"

sleemanj, 2017-07-04:
@SpenceKonde Wow, you are right, the default avrdude.conf doesn't support Tiny 48.. I was going to point the finger at Arduino but... actually it's not in the official avrdude.conf either.  Maybe you should attempt (!) to get that conf data (and any other missing ones) added to at least the Arduino distributed avrdude.conf, not sure which repo it resides in though if any.

beicnet, 2017-07-04:
One suggestion about naming the programmer @SpenceKonde "Arduino/Leo as ISP (ATtiny)" can be maybe "Arduino as ISP (ATtiny)" or "Arduino as ISP (ATtinyCore)"

per1234, 2017-07-04:
>not sure which repo it resides in though if any.

@sleemanj here it is:
https://github.com/arduino/avrdude-build-script
Though it would probably be better to submit a bug report/patch upstream:
http://www.nongnu.org/avrdude/
It's always surprising to me that parts are missing from avrdude.conf.

closed 2017-07-06 by SpenceKonde

SpenceKonde, 2017-07-06:
That update I pushed fixed the name of Arduino as ISP, and changed from ATtiny to ATTinyCore

beicnet, 2017-07-06:
Thank you @SpenceKonde I already downloaded and manually installed, all working as expected! ;)