SpenceKonde/ATTinyCore #151: Support for Arduino IDE 1.8.3

yzubkov, 2017-07-17:
I have installed Arduino IDE 1.8.3 from Arch repos (avr-gcc 7.1.0-1, arduino-avr-core 1.6.19-1, ...), and then followed the instructions to add support for ATTinyCore. However when validating the sketch I get the following errors:

`fork/exec {runtime.tools.avr-gcc.path}/bin/avr-g++: no such file or directory
Error compiling for board ATtiny25/45/85.`

I did check and `avr-g++` file is present under /usr/bin
Curiously, my `platform.txt` file does not contain any reference to the `runtime.tools.avr-gcc.path`. It is just not mentioned in that file anywhere.

Would be nice to know if someone ran into this issue, or if anyone had success running ATtiny sketches under the Arduino IDE 1.8.3. Thanks!

per1234, 2017-07-17:
Try downloading and installing the Arduino IDE from:
http://www.arduino.cc/en/Main/Software
The Linux package managers like to muck around with the IDE.

`runtime.tools.avr-gcc.path` is automatically generated by the IDE based on avr-gcc being installed in a specific location. Having it in /usr/bin is probably the problem. You may need to let the Arduino IDE have it's own copy of the compiler where it wants it and then if you want avr-gcc 7.1.0-1 installed separately for use outside the Arduino IDE that's up to you.

I can compile fine for ATtiny25/45/85 using Arduino IDE 1.8.3

yzubkov, 2017-07-17:
Thanks for your help, I was able to resolve everything by using the official download.

closed 2017-07-17 by yzubkov