SpenceKonde/ATTinyCore #152: 8MHz Prescaler for ATTiny84?

abek42, 2017-07-21:
Hi,

I uploaded the following blink sketch to an ATTiny84 with an external 8MHz Crystal Oscillator.

```
void setup() {
  //Pin 12 on the ATTiny84
  pinMode(1, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(1, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(1, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```
If I select the Oscillator option as 1MHz internal, I can see blips at 1s intervals.
However, any 8MHz option (internal/external) the blip interval is 8s.

I believe some configuration setting needs updating (I am encountering the same problem if I use the damellis core too)

Programmer config:
IDE: 1.8.2, Latest ATTinyCore, Uno as ISP

Thanks.

SpenceKonde, 2017-07-21:
You forgot to do "burn bootloader" to set the fuses to use the selected clock setting - the chip is running at 1mhz because it still has the default fuse settings on.  "Burn bootloader" to set the fuses with the desired clock option selected, and then it will run at the correct speed.

closed 2017-07-21 by SpenceKonde

abek42, 2017-07-21:
That fixed it! Do you mind adding this line to your Readme.md reminding a user to do so?

Thanks!

SpenceKonde, 2017-07-21:
I'm pretty sure I have mentioned this - where do you think it needs to go to help make it clearer?

srw777, 2017-07-21:
This should be in the first section of the readme.md file. (or in the bootloader section?)  This might be obvious to people who have been using the core for a while, but it took me quite a while to figure this out as well.

SpenceKonde, 2017-07-21:
Added a note at the top about this

abek42, 2017-07-22:
I would recommend adding a note in the "Supported clock speeds:" section saying, "NOTE: Selecting any of these speeds through the Arduino IDE doesn't set the correct fuses on the chip automatically. After changing speed settings, use the "Burn Bootloader" option to set the fuses correctly"

The updated version reads well but the above is for real beginners.