SpenceKonde/ATTinyCore #155: ATtiny85 Tone() doesn't work at all

zamanturja, 2017-07-22:
is there any other process to solve the problem?

SpenceKonde, 2017-07-22:
I have definitely been able to get tone() to work on attiny85. I don't have one handy, but it definitely worked. It is known that it doesn't work when you repeatedly call tone() in a tight loop , but I definitely got tones out of it...

zamanturja, 2017-07-22:
i have the board manager from this site: githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json

i have tried with all the pins of ATtiny 85, but still not working. I'm using a minipro programmer to flash the chip using the hex file generated by arduino IDE(processor ATtiny85, Internal 8 MHz selected)...

am i doing it in the right way ?

SpenceKonde, 2017-07-22:
You're not using this core!
Tone is busted on the d. a. mellis core. 

Board manager url for this core is http://drazzy.com/package_drazzy.com_index.json

SpenceKonde, 2017-07-22:
It should definitely work on this core (note that pin 4 will give best results, since it can be done fully in hardware)

closed 2017-07-22 by SpenceKonde

zamanturja, 2017-07-22:
Thanks a lot, you solved it !