SpenceKonde/ATTinyCore #156: Using ATtiny841 USARTs under interrupt

technoblogy, 2017-07-23:
I don't seem to be able to use the ATtiny841 USARTs under interrupt. When I define an interrupt service routine with a plausible name:

```
ISR (USART0_RXC_vect) {     
  ...
}
```
I get the error:

/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_modified_sketch_104319/I2CGPS2.ino:179:6: warning: 'USART0_RXC_vect' appears to be a misspelled signal handler, missing __vector prefix [-Wmisspelled-isr]

Any suggestions?

SpenceKonde, 2017-07-24:
That is a warning not an error. I would expect you'd get an error about using the same ISR twice if you tried that though, because the core itself already defines that ISR - the core already implements serial functionality on both serial ports - you can access them with Serial and Serial1 (they work like they do on normal Arduino's)

technoblogy, 2017-07-24:
Ah - thanks.

closed 2017-07-24 by technoblogy