SpenceKonde/ATTinyCore #158: Verification error uploading to ATtiny1634

technoblogy, 2017-08-06:
I've assembled one of your ATtiny1634 breakout boards (Rev. C) with a minimum set of components (ATtiny1634R, crystal, C1, C2, and C4) and am trying to program it via ISP using the Sparkfun Tiny Programmer. I can set the fuses OK, but when I upload Blink I get:

avrdude: verification error, first mismatch at byte 0x0002
         0x40 != 0x50

The settings are:

ATtiny1634 (No bootloader)
BOD: Disabled
LTO: Disabled
Clock: 8MHz (external)
Programmer: USBtinyISP (ATTinyCore)

Any suggestions of things I can try?

SpenceKonde, 2017-08-06:
Can you enable verbose uploads and try again, and post the output? Worth a
try, though figuring out content mismatch errors is often really hard.

It's worth noting that a lot of people have trouble with the usbtinyisp
(Not just with this core). People seem to have much better luck with the
usbasp, of the popular cheap ISP programmers. It's not clear why peoples'
experiences vary so much with them (I'm suspicious that some of the ones
making the rounds are bad). Do you know that your programmer works on other
AVRs?

-Spence

technoblogy, 2017-08-08:
Thank you - I'll investigate and report back.

technoblogy, 2017-08-08:
Here's the verbose output:

```
/Applications/Arduino 1.8.2.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/avrdude.conf -v -pattiny1634 -cusbtiny -Uflash:w:/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_346547/Blink.ino.hex:i 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/avrdude.conf"
         User configuration file is "/Users/david/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 5.0
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 020:007
         AVR Part                      : ATtiny1634
         Chip Erase delay              : 15000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no        256    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     16384   32    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Setting SCK period to 5 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9412 (probably t1634)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: Setting SCK period to 5 usec
avrdude: reading input file "/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_346547/Blink.ino.hex"
avrdude: writing flash (970 bytes):

Writing | ################################################## | 100% 0.75s

avrdude: 970 bytes of flash written
avrdude: verifying flash memory against /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_346547/Blink.ino.hex:
avrdude: load data flash data from input file /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_346547/Blink.ino.hex:
avrdude: input file /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_346547/Blink.ino.hex contains 970 bytes
avrdude: reading on-chip flash data:

Reading | ##############################################An error occurred while uploading the sketch
#### | 100% 1.74s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0002
         0x40 != 0x50
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

SpenceKonde, 2017-08-11:
It looks like problems with verification errors are not uncommon with the USBTinyISP, but my brief investigation didn't lead me to a clear explanation of why, or whether it's possible to modify the core to fix it. Nobody seems to be able to agree on either the root cause, or even what triggers it, and I don't have a USBTinyISP to investigate with (it seems that there are many implementations of the USBTinyISP design, so that might matter too) myself. 
So at this point, I don't really know what to do. Can anyone else who has experience with USBTinyISP help out here? 

As a workaround/in the meantime, I would suggest trying another programmer; I use the USBAsp programmers myself (and that's what the core is tested with).

technoblogy, 2017-08-11:
Thanks - I'll get hold of the USBAsp and try that.

per1234, 2017-08-11:
I've definitely encountered this issue with USBtinyISP but I never spent a lot of time to find a solution because I already have the USBasp that works great. I own two or three different of the cheapest versions of the USBtinyISP available on Aliexpress and haven't found a difference in reliability between the different designs. I'd be happy to do any testing you might need but unfortunately I don't have an ATtiny1634, only ATtiny85, ATtiny84, and ATtiny2313.

technoblogy, 2017-08-16:
A bit more information:

I've got hold of a USBasp and verified that I can program an ATtiny85 with it.

When I use it to do Burn Bootloader to the ATtiny1634R I get:

```
avrdude: set SCK frequency to 187500 Hz
avrdude: set SCK frequency to 187500 Hz
avrdude: set SCK frequency to 187500 Hz
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

Error while burning bootloader.
```
If I try and upload Blink I get:
```
avrdude: set SCK frequency to 187500 Hz
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

the selected serial port 
 does not exist or your board is not connected
```
I've assembled two PCBs so I think it's unlikely to be a component or soldering problem. I assume the fact I chose an ATtiny1634R shouldn't cause a problem?

Any further suggestions?

SpenceKonde, 2017-08-18:
ATTiny1634R is fine - that's what I use on the assembled boards I sell. 

That error almost always indicates a wiring problem. Are you sure the ISP programmer is plugged in the right way around? With the connector plugged in from the top of the board, ground is on the end closer to the chip. 

Also, check for other silly stuff - is the chip soldered on the right way around? Pin1 should face the end with the 6-pin header. Have you checked all connections with a multimeter? (I check continuity, and for shorts to Vcc/Gnd, preferably measuring from the pins on the USBAsp where it goes into the 10-pin connector to the pins on the chip itself)
Are you sure that your soldering technique is working with the SMD boards? Some people have issues with this; The technique I recommend is what they call "drag soldering"; I use generic "no clean gel flux", the kind that comes in the syringe and leaded solder.

technoblogy, 2017-08-19:
I've checked the wiring, and all the soldering, so I'm baffled. I'm sure I'll figure it out, and I'll let you know if I do. Thanks for the help.

SpenceKonde, 2017-08-23:
If you send a good (close up, high res) picture I could see if I see anything that would cause alarm - though over photos I'm not sure how much help this would be. 

Another thing that might help would be to try stepping down a level and using the Arduino as ISP sketch to program, and see if that doesn't work. If it does work, update the firmware on the USBAsp to the PeterVH/bperry 1.06 version.

technoblogy, 2017-08-24:
Thanks for the suggestions. I've got one more thing to try, and then I'll try sending you a photo.

Regards, David

technoblogy, 2017-08-24:
OK, with further tests I've got partial success. I have assembled two of your boards with identical components: ATtiny1634R, 8MHz crystal, 2x30pf capacitors, and 0.1uF across supply, and an LED with series resistor.

**Using the Sparkfun Tiny Programmer**

With the core settings: ATtiny1634 (No Bootloader), BOD disabled, LTO disabled, Clock 8MHz (external), Programmer: USBTinyISP (ATTinyCore), Burn Bootloader works. I've done this on both boards to set the fuses for an 8MHz crystal.

However, Using the Sparkfun Tiny Programmer to try and upload the Blink sketch always fails with something like:

```
avrdude: verification error, first mismatch at byte 0x0002
         0x40 != 0x50
avrdude: verification error; content mismatch
avrdude: verification error; content mismatch
```

**Using the USBASP Programmer**

The core settings are: ATtiny1634 (No Bootloader), BOD disabled, LTO disabled, Clock 8MHz (external), Programmer: USBasp (ATTinyCore).

If I open the Blink sketch, plug in the USBASP,  and do Upload, the program uploads and runs OK.

If I do Upload a second time, it fails with the error:

```
avrdude: set SCK frequency to 187500 Hz
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

the selected serial port 
 does not exist or your board is not connected
```
If I unplug the USBASP from the USB port on my computer and plug it back in again, Upload now works.

If I try to do Burn Bootloader to set the fuses, it always fails with the above error message, even when the USBASP is first plugged in.

I haven't successfully uploaded the Optiboot bootloader under any conditions.

Hope this gives you some ideas about what might be going wrong.

Thanks, David

sleemanj, 2017-08-25:
> avrdude: set SCK frequency to 187500 Hz

What is the full avrdude command line for the USBasp?

Reason I ask is it is unexpected for the Arduino IDE to cause avrdude to try and set a specific sck frequency (sck period, set through the -B option to avrdude) as it does not typically provide that option on the avrdude command line.  Current avrdude (no idea on older ones) will default to automatic sck and you would get a message like

> avrdude: auto set sck period (because given equals null)

technoblogy, 2017-09-03:
Sorry for the delay in replying.

The command is:

`
/Applications/Arduino 1.8.2.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/avrdude.conf -v -pattiny1634 -cusbasp -Uflash:w:/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_146201/Blink.ino.hex:i 
`

boddeke, 2017-09-06:
i got a similar issue: the attiny1634 seems to get programmed badly. below the hex and the flash dump read back after programming (using avrdude interactively). the first 2 "lines" are almost good, then a couple line not programmed, a line completely off, again unprogrammed area and at 0x100 again 2 good lines. this is so weird.

the flash memory contents is always the same. not randomly wrong. programming always fails with:

avrdude: verification error, first mismatch at byte 0x0002
         0x08 != 0x38

anyone any idea??

thanks,
frank

main.hex
```
:100000000C9438000C944A000C944A000C944A005A
:100010000C944A000C944A000C944A000C944C0036
:100020000C944A000C944A000C944A000C944A0028
:100030000C944A000C944A000C944A000C944A0018
:100040000C944A000C944A000C944A000C944A0008
:100050000C944A000C944A000C944A000C944A00F8
:100060000C944A000C944A000C944A000C944A00E8
:1000700011241FBECFEFD4E0DEBFCDBF21E0A0E052
:10008000B1E001C01D92A430B207E1F70E947100F7
:100090000C9489000C9400001F920F920FB60F92DF
:1000A00011248F939F93AF93BF9380910001909100
:1000B0000101A0910201B09103010196A11DB11DA2
:1000C0008093000190930101A0930201B09303017A
:1000D000BF91AF919F918F910F900FBE0F901F9086
:1000E0001895869A859A409A8D9A2FEF80E792E02C
:1000F000215080409040E1F700C000008D982FEF24
:1001000088EF95E1215080409040E1F700C0000069
:06011000EBCFF894FFCFD5
:00000001FF
```

```
>>> dump flash 0 16384 
0000  0c 94 08 00 0c 94 4a 00  0c 94 4a 00 0c 94 4a 00  | ... .J. .J. .J.|
0010  00 04 0a 00 0c 84 40 00  0c 94 48 00 00 80 00 00  |.. . .@. .H.....|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  00 00 00 00 00 92 00 00  80 02 00 01 00 00 00 00  |................|
0090  00 00 80 00 00 00 00 00  00 00 00 00 01 10 01 00  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0100  88 ef 95 e1 21 50 80 40  90 40 e1 f7 00 c0 00 00  |....!P.@.@......|
0110  eb cf f8 94 ff cf ff ff  ff ff ff ff ff ff ff ff  |................|
0120  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0130  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0140  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

```

boddeke, 2017-09-06:
solved my problem! seems my version of the avrdude.conf is wrong for attiny1634. using the attached works for me.

[avrdude.conf.zip](https://github.com/SpenceKonde/ATTinyCore/files/1282485/avrdude.conf.zip)

technoblogy, 2017-09-07:
That is the version of avrdude.conf that I have been using, and I am definitely having problems.

Please can you confirm whether you can successfully upload the Optiboot bootloader under any conditions.

boddeke, 2017-09-07:
no boot loader here. just, i thought, the same problem.

maybe you can use avrdude interactively to debug the problem? i just compared the hex file agains what i read back from the micro.

SpenceKonde, 2017-09-11:
Okay let's get back to first principles. Does it work with Arduino as ISP? Arduino as ISP is a miserable programmer for a number of reasons, but assuming all the connections are made correctly (this seems to be something people have a really hard time with, judging from discussion on the forums) and reset is disabled and all caveats are accounted for, I've never heard of it not working. 

What firmware do you have loaded on the USBAsp? I'm using the PeterVH/bperrypap 1.06 firmware. I have heard of some USBAsps coming pre-loaded with questionable firmware.

technoblogy, 2017-09-11:
Thank you for pursuing this. I'll do the tests you've suggested and get back to you.

Regards, David

luigizuccarelli, 2017-09-19:
@technoblogy - I had exactly the same problems. I tried with a home made USB attiny85 programmer I made. It worked with the 85 and then on the 1634 - with the verification failing on the 0x0002 byte. I then used a pi zero with linuxgpio I got the exact same error. I then proceeded to try an external osc (8Mhz) but  with the same results. I figured it then had to be with the config. I tried @boddeke's conf and all worked, with and without external osc - thanks @boddeke :+1:

technoblogy, 2017-09-19:
Please can you give a link to @boddeke's conf.

boddeke, 2017-09-19:
it is already there, a few posts up (avrdude.conf.zip)

technoblogy, 2017-09-19:
Please can you confirm whether you can successfully upload the Optiboot bootloader using @boddeke's avrdude.conf file (version 1371 2016-02-15).

technoblogy, 2017-09-20:
**@SpenceKonde** as suggested I have tried using Arduino as ISP, and here are the results. These were all obtained with a vanilla copy of ATTinyCore (Latest commit 3d647b4  8 days ago):

**Test 1 - using ATtiny1634 (No bootloader) option with programmer: Arduino as ISP (ATTinyCore)**

Settings:

ATtiny1634 (No bootloader)
BOD Disabled
LTO Disabled
Clock: 8MHz (internal)
Programmer: Arduino as ISP (ATTinyCore)

Upload: fails:

```
avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x01
avrdude: stk500_initialize(): (a) protocol error, expect=0x14, resp=0x10
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): unknown response=0x12
the selected serial port avrdude: stk500_disable(): unknown response=0x12
 does not exist or your board is not connected
```

Burn Bootloader: fails:

```
avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x01
avrdude: stk500_initialize(): (a) protocol error, expect=0x14, resp=0x10
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): unknown response=0x12
Error while burning bootloader.
```

**Test 1 - using ATtiny1634 (No bootloader) option with programmer: Arduino as ISP**

Settings:

ATtiny1634 (No bootloader)
BOD Disabled
LTO Disabled
Clock: 8MHz (internal)
Programmer: Arduino as ISP

Upload: Works reliably.

Burn Bootloader: failed once:

 ```
***failed;  
avrdude: WARNING: invalid value for unused bits in fuse "efuse", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0x07 instead of 0xff (double check with your datasheet first).
```

Tried again: seems to work reliably now.

**Test 3 - using ATtiny1634 (No bootloader) option with programmer: Arduino as ISP**

Settings:

ATtiny1634 (Optiboot)
BOD Disabled
LTO Disabled
Clock: 8MHz (external)
Programmer: Arduino as ISP

Burn Bootloader: worked: (LED flashes twice every second)

From looking at the verbose upload output it seems that the difference between **Arduino as ISP** and **Arduino as ISP (ATTinyCore)** is that **Arduino as ISP** uses the conf file "/Applications/Arduino 1.8.2.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf", which has the same version number as @boddeke's avrdude.conf file referenced in earlier posts, and **Arduino as ISP (ATTinyCore)** uses "/Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/avrdude.conf", the one in the ATTinyCore. Is this correct?

SpenceKonde, 2017-09-20:
The fuck? 
avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

SpenceKonde, 2017-09-20:
Also what version of the official AVR boards do you have installed? Just for completeness sake.

technoblogy, 2017-09-20:
It's 1.6.18.

technoblogy, 2017-09-20:
So would you expect both **Arduino as ISP** and **Arduino as ISP (ATTinyCore)** to work? What's the difference?

SpenceKonde, 2017-09-20:
Difference is which avrdude.conf file they use (attinycore) uses the one supplied with the core. The other one uses the one that comes with the default AVR board package. The one that comes with the default AVR board package throws warnings about the way some of the fuses specified handle unused bits, and is missing entries (or correct ones) for several of the more obscure attinys.

SpenceKonde, 2017-09-20:
If you could run those tests with verbose upload enabled and post the command that avrdude is being called with (first line of the verbose output), that would help me investigate (especially with my limited time in the lab lately). 

That's a really strange issue. What are you using as the Arduino as ISP? Did you remember to disable the autoreset on the arduino being used as ISP?

technoblogy, 2017-09-20:
It's an original Arduino Uno R3. I didn't know about needing to disable autoreset - I've never had to do that before - why, and how do I do it?

I'll run the tests again and post verbose output, although I won't be able to do that until tomorrow.

SpenceKonde, 2017-09-20:
10uF cap between Reset and Ground on the arduino acting as ISP programmer. It's needed to ensure there aren't problems caused by the autoreset of the board when the serial port is opened to start using it as ISP.  This may be timing sensitive - some people say they don't need it, but most people say they do.

technoblogy, 2017-09-21:
**@SpenceKonde** here is the same sequence of tests as in my earlier post, but with verbose upload output:

**Test 1 - using ATtiny1634 (No bootloader) option with programmer: Arduino as ISP (ATTinyCore)**

Settings:

ATtiny1634 (No bootloader)
BOD Disabled
LTO Disabled
Clock: 8MHz (internal)
Programmer: Arduino as ISP (ATTinyCore)

Upload: fails:

```
/Applications/Arduino 1.8.2.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/avrdude.conf -v -pattiny1634 -carduino -P/dev/cu.usbmodem1411 -b19200 -Uflash:w:/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_133437/Blink.ino.hex:i 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/avrdude.conf"
         User configuration file is "/Users/david/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1411
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny1634
         Chip Erase delay              : 15000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no        256    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     16384   32    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: -1558273796
         Firmware Version: -1073745304.444305
         Topcard         : STK502
         Vtarget         : 1.8 V
[... 548 lines cut ...]
Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9412 (probably t1634)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/optiboot/optiboot_attiny1634at8.hex"
avrdude: writing flash (16384 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 16384 bytes of flash written
avrdude: verifying flash memory against /Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/optiboot/optiboot_attiny1634at8.hex:
avrdude: load data flash data from input file /Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/optiboot/optiboot_attiny1634at8.hex:
avrdude: input file /Users/david/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/optiboot/optiboot_attiny1634at8.hex contains 16384 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 16384 bytes of flash verified

avrdude done.  Thank you.

```

eugenelivet, 2017-11-22:
Hello, I have a similar problem. 

I programmed many ATtiny1634 without any problem few months ago now. Actually, I need to use it again on new projects but I have many problems to program them.

I'm using Arduino 1.8.5 and ATtinycore 1.1.4
I programmed 8MHz internal bootloader with a AVRISP MKII.

Below, the verbose output :

```
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/regisboullard/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.4/avrdude.conf"
         User configuration file is "/Users/regisboullard/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-A50285BI
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny1634
         Chip Erase delay              : 15000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no        256    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     16384   32    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x90
         Hardware Version: 3
         Firmware Version: 6.444305
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9412 (probably t1634)
avrdude: reading input file "/var/folders/3q/k8wxkc617931kmc7mm11b3gc0000gn/T/arduino_build_533658/Blink.ino.hex"
avrdude: writing flash (2198 bytes):

Writing | #avrdude: loadaddr(): (b) protocol error, expect=0x14, resp=0x90
####avrdude: loadaddr(): (b) protocol error, expect=0x14, resp=0x90
#######################avrdude: loadaddr(): (b) protocol error, expect=0x14, resp=0x90
#############avrdude: loadaddr(): (b) protocol error, expect=0x14, resp=0x90
######### | 100% 2.22s

avrdude: 2198 bytes of flash written
avrdude: verifying flash memory against /var/folders/3q/k8wxkc617931kmc7mm11b3gc0000gn/T/arduino_build_533658/Blink.ino.hex:
avrdude: load data flash data from input file /var/folders/3q/k8wxkc617931kmc7mm11b3gc0000gn/T/arduino_build_533658/Blink.ino.hex:
avrdude: input file /var/folders/3q/k8wxkc617931kmc7mm11b3gc0000gn/T/arduino_build_533658/Blink.ino.hex contains 2198 bytes
avrdude: reading on-chip flash data:

Reading | 
avrdude: stk500_paged_load(): (a) protocol error, expect=0x10, resp=0x88
avrdude: stk500_cmd(): protocol error
avr_read(): error reading address 0x0001
    read operation not supported for memory "flash"
avrdude: failed to read all of flash memory, rc=-2

avrdude done.  Thank you.

le port série sélectionné 
 n'existe pas ou votre Arduino n'est pas connectée
```
Thanks for any help.

MCUdude, 2018-02-17:
I think there's something wrong with the avrdude.conf option for ATtiny1634. Burning bootloader using USBtinyISP didn't work for me either. However it _did_ work burning bootloader using USBtinyISP after using [this avrdude option](http://www.avrfreaks.net/comment/943976#comment-943976). It's from the same post as the one used with ATTinyCore has it's origin.

Uploading code using the bootloader still doesn't work though. Here's the output:
```
/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino12/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/ATTinyCore/avr/avrdude.conf -v -pattiny1634 -carduino -P/dev/cu.wchusbserial1410 -b115200 -D -Uflash:w:/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_802437/Blink.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/Hans/Documents/Arduino/hardware/ATTinyCore/avr/avrdude.conf"
         User configuration file is "/Users/Hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.wchusbserial1410
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         Setting bit clk period        : 5.0
         AVR Part                      : ATTINY1634
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no        512    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     16384   32    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 6.2
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: stk500_initialize(): n_extparms 4 mayor 6 minor 2 
avrdude: stk500_initialize(): n_extparms = 4
trying to set bitclock at 5.000000
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

An error occurred while uploading the sketch
```

SpenceKonde, 2018-02-18:
!!! 

Thank you, I think I see what may be the cause of the issue. I've also just received a USBTinyISP in the mail, so have something to test it with.

SpenceKonde, 2018-02-23:
Reopening until fix is verified.

MCUdude, 2018-02-23:
I'm away this weekend, but I'll test it on monday.

MCUdude, 2018-02-25:
Just pulled the latest version of ATTinyCore.

My USBasp has a default SPI clock of about 350kHz, so I needed to use a slow programmer option in the IDE (with -B32). Was able to set fuses and and burn bootloader without any issues.

Don't know the ISP clock frequency on my USBtinyISP, but I was able to set fuses without any errors. However I wasn't able to burn the bootloader. Here's the error:

```
avrdude: verifying ...
avrdude: 16384 bytes of flash verified

avrdude done.  Thank you.

avrdude(27785,0xa6f4e1c0) malloc: *** error for object 0x806808: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
Error while burning bootloader.
```

No errors where returned when burning bootloader using USBtinyISP with -B32. No idea why though.. I use this USBtinyISP for everything, and I've never seen this error? I believe it may be the hex file, avrdude.conf or the hardware itself.

However I was never actually able to upload anything using the bootloader. I get the same error whenever I use the USBtinyISP or USBasp. Here's the error:

```
/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino12/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/ATTinyCore/avr/avrdude.conf -v -pattiny1634 -carduino -P/dev/cu.wchusbserial1410 -b57600 -D -Uflash:w:/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_615287/Blink.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/Hans/Documents/Arduino/hardware/ATTinyCore/avr/avrdude.conf"
         User configuration file is "/Users/Hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.wchusbserial1410
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny1634
         Chip Erase delay              : 9000 us
         PAGEL                         : PB3
         BS2                           : PB1
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no        256    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     16384   32    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 6.2
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: stk500_initialize(): n_extparms 4 mayor 6 minor 2 
avrdude: stk500_initialize(): n_extparms = 4
trying to set bitclock at 5.000000
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

An error occurred while uploading the sketch
```

I'm compiling on Arduino IDE 1.8.5 for MacOS

SpenceKonde, 2018-02-28:
I have reproduced the issue with a USBTinyISP that I got, and confirmed that this fixed it. 

@MCUdude - Not sure why you're having that problem - I've never seen that error message when uploading via serial using the bootloader....

closed 2018-02-28 by SpenceKonde

MCUdude, 2018-02-28:
> Not sure why you're having that problem - I've never seen that error message when uploading via serial using the bootloader....

Me neither. I think you should actually try to figure out why it doesn't work for me. Is it a Mac OSX issue? Maybe someone else with a mac can give this a try? I'm using avrdude shipped with IDE 1.8.5.

SpenceKonde, 2018-02-28:
Other people on forum with 1.8.5 on macos post avrdude output listing version as: 
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35

Yours shows:
avrdude: Version 6.3-20171130
And the path:
/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino12/bin/avrdude - that arduino12 bit doesn't look normal.

It looks like you've got a special/different version of avrdude...... The version of AVRdude (and the toolchain) used by attinycore are the same as whatever the arduino avr core uses - have you updated that to some strange experimental version, or done something else that updates avrdude?

MCUdude, 2018-02-28:
You're right, I've been using the latest experimental version of Arduino AVR boards (1.6.206). I've been using it because it support the latest ATmega*PB devices. When using the Avrdude version that ships with IDE 1.8.5, everything works fine. Seems like there's something wrong with Avrdude then..

```
$ /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/ATTinyCore/avr/avrdude.conf -v -pattiny1634 -carduino -P/dev/cu.wchusbserial1410 -b115200 -Uflash:w:/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_808372/Blink.ino.hex:i 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/Hans/Documents/Arduino/hardware/ATTinyCore/avr/avrdude.conf"
         User configuration file is "/Users/Hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.wchusbserial1410
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny1634
         Chip Erase delay              : 9000 us
         PAGEL                         : PB3
         BS2                           : PB1
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no        256    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     16384   32    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 6.2
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9412 (probably t1634)
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_808372/Blink.ino.hex"
avrdude: writing flash (1044 bytes):

Writing | ################################################## | 100% 1.77s

avrdude: 1044 bytes of flash written
avrdude: verifying flash memory against /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_808372/Blink.ino.hex:
avrdude: load data flash data from input file /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_808372/Blink.ino.hex:
avrdude: input file /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_808372/Blink.ino.hex contains 1044 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.60s

avrdude: verifying ...
avrdude: 1044 bytes of flash verified

avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

I should probably create an issue over at the Arduino toolchain repo.

commit 63b92809 (2018-02-23): Possibly, maybe, fix #158?! 

Please help me out testing this - was this all it took?

A better question is how it ever worked with any programmer, if I understand what was going on here.