SpenceKonde/ATTinyCore #160: SPI library for ATtiny1634

alessio31183, 2017-08-17:
is there something to support the SPI communication for the Attiny1634?
I didn't find any library to use the SPI con this uC.

Thanks

SpenceKonde, 2017-08-17:
It is supplied as part of the core. Just include SPI.h like it was a normal
arduino - and it will use the USI for SPI. There are a few notes on the
universal SPI library in the documentation - among other things, the clock
speed option isn't yet supported, but for typical applications it should
just work.

alessio31183, 2017-08-17:
And then in my driver constructor:
LIS3DHH::LIS3DHH(byte clockDiv, byte ssPin, byte mode)
{
  pinMode(ssPin,OUTPUT);
  // Ensure the RAM chip is disabled in the first instance
  disable();
  SPI.beginTransaction(SPISettings(clockDiv, MSBFIRST, mode));

  _ssPin = ssPin;

  // Set the RAM operarion mode flag as undefined
  _current_mode = mode;
}

I cannot use the SPISettings(clockDiv, MSBFIRST, mode). Right?

How can I fix this and set the SPI settings correctly?

Thanks

alessio31183, 2017-08-17:
Sorry but shall I have to replace my avr folder in arduino-1.8.1/hardware/arduino with your folder?

alessio31183, 2017-08-17:
Sorry but shall I have to replace my avr folder in arduino-1.8.1/hardware/arduino with your folder?

I cannot understand how the system know which SPI.h to take, yours or the original one.

alessio31183, 2017-08-17:
Ok , solved updating the library.

It was an old version

Thanks

SpenceKonde, 2017-08-17:
Glad you figured it out - is this sorted out then (ie, can I close this)?

alessio31183, 2017-08-18:
Yes sure.
Thanks again

closed 2017-08-23 by SpenceKonde

colesnicov, 2018-05-24:
Hi guys. I've got the problem:

```

Compiling library "SPI"
"/home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -Wall -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp" -o "/tmp/arduino_build_431784/libraries/SPI/SPI.cpp.o"
In file included from /home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/avr/include/avr/io.h:99:0,
                 from /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.h:27,
                 from /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:19:
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp: In static member function 'static void tinySPI::begin()':
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern/pins_arduino.h:126:21: error: 'DDC1' was not declared in this scope
 #define USCK_DD_PIN DDC1
                     ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:263:25: note: in expansion of macro 'USCK_DD_PIN'
     USI_SCK_PORT |= _BV(USCK_DD_PIN);   //set the USCK pin as output
                         ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern/pins_arduino.h:127:19: error: 'DDB2' was not declared in this scope
 #define DO_DD_PIN DDB2
                   ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:264:25: note: in expansion of macro 'DO_DD_PIN'
     USI_DDR_PORT |= _BV(DO_DD_PIN);     //set the DO pin as output
                         ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern/pins_arduino.h:128:19: error: 'DDB1' was not declared in this scope
 #define DI_DD_PIN DDB1
                   ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:265:26: note: in expansion of macro 'DI_DD_PIN'
     USI_DDR_PORT &= ~_BV(DI_DD_PIN);    //set the DI pin as input
                          ^
Použití knihovny SPI ve verzi 1.0 v adresáři: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI 
exit status 1
Nastala chyba při kompilaci u desky ATtiny1634 (Optiboot).
```