SpenceKonde/ATTinyCore #161: Modern core missing digitalPinToAnalogInput or similar

matthijskooijman, 2017-08-17:
In the regular Arduino core, both analog channel numbers and "digital" pin numbers can be passed to `analogRead()`. In the tinymodern core, I believe you can only pass analog channel numbers (presumably since these two ranges are not equal, but do overlap, making it impossible to distinguish them). Since you often know the pin number to use, but not necessarily the ADC channel corresponding to it, some automated way to translate them would be convenient.

I did find `analogInputToDigitalPin()` in the tinymodern core, but that does the reverse. I now hardcoded the analog channel and used a static_assert to verify it matches the pin number I expect, but just defining a single pin number and automatically translating to a channel number would be more convenient.

SpenceKonde, 2017-08-23:
On the official core, you can either pass the analog channel number (as a number), or the pin number expressed in A# format (eg, A1 for ADC channel 1). They've got the pin numbers chosen so that the A# constants are the same as those digital pins. The ADC pin locations are more complicated on the tinies, so I wasn't able to adopt that scheme - I also detest the Arduino IDE separating pins into "analog" and "digital" pins; this causes a great deal of confusion, which wouldn't be the case if the Arduino IDE made it clear that all pins are digital pins, and a few of them just have the added function that they can also read analog voltages. 

I couldn't figure out a way to make the ADC channel numbering work for the tinies without either renumbering all the pins (which is off the table, as people have code written for the existing pin mapping), or breaking support for using the differential ADC channels (which you can do by passing numbers from table 16-3 (pg 144) in datasheet to analogRead).

You can now use the A# notation for digitalRead()/write() and pinMode (as of latest release), but you can't do the reverse and use digital pin numbers for analogRead. (this is implemented by using the high bit of the A# defines to mark them as analog pins, eg, A0 is 128 0x1000000, and analogRead strips out the high bit, while the digital functions detect it and convert it to the digital pin number with analogInputToDigitalPin(). 

If you can figure out a way to do it without breaking things, it's all ears, but I wasn't able to.

On the smaller issue of an digitalPinToAnalogInput - a macro like that could be made, but that doesn't seem to be a standard macro (searched arduino core and didn't find it). Is it known by a different name on the official core?

SpenceKonde, 2017-09-11:
I'm disinclined towards implementing a non-standard macro here, since as far as I can tell this isn't causing widespread hardship (nobody else has commented on this, and no reply from OP). If there's a standard solution, or it turns out that this is inconveniencing a lot of people, please feel free to reopen and we can revisit this.

closed 2017-09-11 by SpenceKonde

matthijskooijman, 2017-09-11:
W00ps, this fell off my list.

> On the official core, you can either pass the analog channel number (as a number), or the pin number expressed in A# format (eg, A1 for ADC channel 1). They've got the pin numbers chosen so that the [...]

Yeah, that indeed sucks.

> You can now use the A# notation for digitalRead()/write() and pinMode (as of latest release), but you can't do the reverse and use digital pin numbers for analogRead. (this is implemented by using the high bit of the A# defines to mark them as analog pins, eg, A0 is 128 0b1000000, and analogRead strips out the high bit, while the digital functions detect it and convert it to the digital pin number with analogInputToDigitalPin().

Ah, I see. In that case, I think my request might be to make the [`PIN_A0`](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tinymodern/core_pins.h#L39), `PIN_B0`, etc. macros use this same high bit, so those can be passed as-is to both `analogRead` and `digitalRead` (etc) as well. That would certainly help for my usecase, though it might break compatibility (if people would rely on the values of these macros and compare them with raw pin numbers somehow).

> On the smaller issue of an digitalPinToAnalogInput - a macro like that could be made, but that doesn't seem to be a standard macro (searched arduino core and didn't find it). Is it known by a different name on the official core?

Nope, there isn't any need, because you can just feed digital pin numbers to `analogRead` directly.