SpenceKonde/ATTinyCore #162: I2C Bootloader for ATtiny841?

berfel, 2017-08-23:
This is an enhancement request. Obviously.
And a plea for information.

My widget connects to the outside world by only I2C after it's potted. So the firmware has to be _ferpect_ or it needs to support firmware updates via a bootloader. As I routinely put bugs in my code; including bugs provided by others, the bootloader would be inescapable if one does not wish to contribute prematurely to landfill.

Alas, Optiboot doesn't support I2C; and I can't see how it would be able to do so with "trivial" changes. The (sleeping) OpenServo project codebase has a t85 I2C bootloader employing its USI (as I understand) as distinct from the t841's slave-TWI; but that's the closest that I could find over the past week.

Are there any current bootloader projects for I2C on ATtiny; esp. those with at least slave I2C hardware?

SpenceKonde, 2017-08-23:
Optiboot is probably not an appropriate starting point for an I2C bootloader. 

I don't think a huge amount of porting effort would be needed to convert an I2C bootloader that uses the USI or a full TWI module to use the slave module instead - it's a matter of looking at each place where they access I2C-related registers, and looking at the relevant sections of the two datasheets and finding the equivalent register and bit for slave I2C. 

In any event, this is well outside the scope of this core - this core is for supporting these chips with the Arduino IDE. There's no mechanism for I2C programming from within the Arduino IDE - I don't have time to take on unpaid work writing and maintaining an I2C bootloader (ofc, if you want to hire me for this work, send me an email - spencekonde@gmail ).

closed 2017-08-23 by SpenceKonde

berfel, 2017-08-25:
Thanks for the advice.