SpenceKonde/ATTinyCore #163: ATTINY13 support?

colesnicov, 2017-09-03:
Hi anyone. It support ATTINY13 chips?

SpenceKonde, 2017-09-03:
No, use MCUdude's MicroCore for the tiny13.

sleemanj, 2017-09-03:
My fork implemented a lot of optimisation work so that the Tiny13,
4,5,9,10 (and x5, x4) will be useful with ATTinyCore, including print,
serial and tone. 

My distribution of my core (the optiboot repository link you posted)
also departs in heavier use of the boards sub-menu system for those
supported chips to enable the user to pick-and-choose various
optimisation types.  My distro also uses my fork of optiboot (hence
where it lives) instead of the ATTinyCore bootloader binaries but that
is a minor difference really in practical terms, of course the
13/4/5/9/10 don't have bootloaders anyway, no room to waste. 

Spence is always welcome to merge any or all of my fork back, but I can
appreciate how that might be a dawnting task to pick through especially
if he's not familiar with those chips. 

I'm not very familiar with the alternative MCUdude core spence refers
too so I can't say why he might feel that is a better choice. 

 

Links:
------
[1] https://github.com/SpenceKonde/ATTinyCore/issues/163
[2]
https://github.com/

closed 2017-09-04 by colesnicov