SpenceKonde/ATTinyCore #165: Problem with I2C repeated start signal

make-ing, 2017-09-05:
Hi Guys,

i have a problem with i2c since i updated my raspberry to the latest raspbian image. The connection with my attiny88 is no longer working. Since my other i2c sensors still work i assume its an ATTinyCore issue. I found this on the raspberry pi firmare github: https://github.com/raspberrypi/firmware/issues/828

It seems something to to with repeated start signal.

I don't have time today to give you more information but i just wanted to open this topic as soon as possible to give people a place to discus.

More informations are comming soon.

SpenceKonde, 2017-09-06:
> this on the raspberry pi firmare github: raspberrypi/firmware#828

The attiny88 has real i2c hardware on it, so it is using the same i2c code
that is used on the '328p (as used in uno/mega/etc) - the part of this
version of the Wire library used to support devices with hardware i2c is a
direct copy of the official library.

Do you get the same behavior with an arduino board based on the '328p?

I am inclined to suspect a problem with the raspi libraries, esp since
there is so much discussion on that issue you linked.

make-ing, 2017-09-06:
@SpenceKonde thank you forthe fast reply. I checked it with an 328p and it works very good. I will now check and see was the difference between the 328p implementation and the attiny88. I'm not that good in low level languanges so maybe you can alos have a look.

Thank you very much for your good work here.

make-ing, 2017-09-06:
After checking with an logic analyzer i figured out i always get the ioerror when the clock is at 100kHz more often the clock is 62,5kHz which seems to work also with the repeated start condition. I think you can close the issue, as i haven not found any difference between arduino implementation and yours.

SpenceKonde, 2017-09-11:
Gee, I would really expect it to be able to work faster than that!

Was the m328 you tested it on clocked at 16mhz? If so, that would explain why the t88 was more troublesome, being clocked at only 8.

closed 2017-09-11 by SpenceKonde