SpenceKonde/ATTinyCore #166: Problem with OCR1A timer on  ATTINY841

Ripa7777, 2017-09-06:
Hi there,

I am using your core 1.1.4 on ATTINY841 successfully with a fairly large program,  except if I add e.g. the below listed test code. This seem to produce a constant 16ms signal on the output pin. 

It does not matter if I change the pin or the OCR1A value. This code works OK on ATMEL Studio 7 (= toggles the led every second). 

What am I missing / doing wrong  here ?

In use: Windows 10, Arduino 1.8.3.

Thanks for your help.

********
This does not work on Arduino  (No bootloader, clock 1Mhz, BOD disabled, LTO disabled):

#include <avr/io.h>
#include <avr/interrupt.h>

void setup() {
  TIMSK1 = 0;  //Disable all interrupts on timer
  TCCR1B |= (1 << CS11) | (1 << CS10); //Set prescaler to 64
  TCCR1B |= (1 << WGM12); //Set (CTC) mode
  OCR1A = 15625;  // CPU Speed 1MHz 1000000/64 = 15625 
  TIMSK1 |= (1 << OCIE1A); //Enable Compare INT
  sei();
}

void loop() {
  DDRA |= (1 << PA3); //Set pin to Output
  PORTA |= (1 << PA3); //Turn the LED on
  while (1); // Run forever
}

ISR(TIMER1_COMPA_vect)
{
  PORTA ^= (1 << PA3); // Toggle
}

***************************************
But the same code works on AS7:

#include <avr/io.h>
#include <avr/interrupt.h>

int main(void)
{
	
	TIMSK1 = 0; //Disable interrupts on timer
	TCCR1B |= (1 << CS11) | (1 << CS10);	//Set prescaler to 64
	TCCR1B |= (1 << WGM12);	 //Set (CTC) mode
	OCR1A = 15625;	  //CPU Speed 1MHz 1000000/64 = 15625
	TIMSK1 |= (1 << OCIE1A); //Enable Compare INT
	sei();

	DDRA |= (1 << PA3); //Set pin to Output
	PORTA |= (1 << PA3); //Turn the LED on	
	while (1); // Run forever
	
}

ISR(TIMER1_COMPA_vect)
{
	PORTA ^= (1 << PA3); // Toggle every second
}

SpenceKonde, 2017-09-11:
The timers are all initialized on startup for output compare PWM before setup() is run (just like on any other Arduino)
Your code assumes that the registers are in their power on configuration - you never reset TCCR1B or TCCR1A to a known state. You need to reset these to zero (or - better still - set them to the value you want them to be in a single operation)

That is, I think you want this: 

`
#include <avr/io.h>
#include <avr/interrupt.h>

void setup() {
TIMSK1 = 0; //Disable all interrupts on timer
TCCR1A=0; //reset to power on state
TCCR1B = (1 << CS11) | (1 << CS10)| (1 << WGM12); //prescaler 64, CTC mode
OCR1A = 15625; // CPU Speed 1MHz 1000000/64 = 15625
TIMSK1 |= (1 << OCIE1A); //Enable Compare INT
sei();
}

void loop() {
DDRA |= (1 << PA3); //Set pin to Output
PORTA |= (1 << PA3); //Turn the LED on
while (1); // Run forever
}

ISR(TIMER1_COMPA_vect)
{
PORTA ^= (1 << PA3); // Toggle
}
`

In the future, when you have a problem with an on-chip peripheral, the first thing I would do is print the values of the relevant registers to serial. The '841 has a real serial port (two of them, in fact), so getting debugging info out is just as easy as a normal Arduino.

closed 2017-09-11 by SpenceKonde

Ripa7777, 2017-09-11:
Dr. Azzy, with your short response I learned more than with those long evenings 
I have been trying to comprehend how the chip functions. The serial hint made my
day too (being newbie).

Thank you for your help ! 

R