SpenceKonde/ATTinyCore #168: upload fails for ATTinyCore ATtiny85 & ATtiny84 with sloeber

jantje, 2017-09-09:
Hi I'm Jantje from the arduino eclipse plugin named sloeber https://github.com/Sloeber/arduino-eclipse-plugin
I got a issue report that the upload failed with sloeber but workes fin inArduino IDE. After investigation I don't understand where your package defines for id attinyx5
Arduino ide uses the following values which seem to  work
upload.speed=19200
upload.protocol=stk500v1
I have no clue where Arduino IDE gets these from.
Can you help me out here?

Here is a link to the sloeber issue upload fails for ATTinyCore ATtiny85 & ATtiny84 https://github.com/Sloeber/arduino-eclipse-plugin/issues/810

SpenceKonde, 2017-09-11:
What programmer is being used? The settings used for uploading should depend on the selected programmer (and are pulled from programmers.txt). Those look like the settings for Arduino as ISP. This core uses the upload.protocol=arduino for arduino as ISP, which as I understand is almost identical to STK500 but with a fix to some corner case that's relevant in arduino uploads. I don't recall the details on that though.

SpenceKonde, 2017-09-18:
Issue in Sloeber marked as fixed. 

Closing - thanks for making the effort to support this core on Sloeber!

jantje, 2017-09-18:
I still hardcoded the values :-( 
So though the issue is fixed I still don't know where the  values come from.
So other boards (needing different values) may fail

jantje, 2017-09-18:
I searched for cstk500v1 in programmers.txt externalprogrammers.txt in ATTinycore and the arduino avr programmers (version 1.6.20) and I find nothing.

SpenceKonde, 2017-09-18:
When you uploaded and saw the IDE use that value, I suspect you had Arduino as ISP selected, instead of Arduino as ISP (ATTiny) - that is, you had the programmer definition from the official arduino core selected instead of the one associated with my core selected. 

From https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/programmers.txt

`arduinoasisp.name=Arduino as ISP
arduinoasisp.communication=serial
arduinoasisp.protocol=stk500v1
arduinoasisp.speed=19200
arduinoasisp.program.protocol=stk500v1
arduinoasisp.program.speed=19200
arduinoasisp.program.tool=avrdude
arduinoasisp.program.extra_params=-P{serial.port} -b{program.speed}`

the c at the start comes from platform.txt, specifically this line:

`tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} {upload.verify} -p{build.mcu} -c{upload.protocol} "-P{serial.port}" -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"`

(Now, you may be wondering WHY I needed to duplicate the programmers in this core - that's due to a bug/feature with the IDE that causes it to use the default core's avrdude.conf if the programmer used is one supplied with the default arduino avr core. Unfortunately, that version of avrdude.conf doesn't support most of the attiny chips. I've raised this issue with the development team, but some other developers of third party addons said that they relied on this behavior. )

jantje, 2017-09-18:
So I looked for cstk500v1 instead of stk500v1 (facepalm)

>(Now, you may be wondering WHY I needed to duplicate the programmers in this core

This whole referencing thing hasn't been thought through properly. runtime.platform.path should point to where the board was selected and not where it references to.
I'm currently using referencing.XXX.path and referenced.XXX.path.
XXX can be upload, core ...

But I see it now and with the changes I did in another issue (that add both referenced platform and referencing platfor programmers.txt) I think this issue can be closed.
Thanks

closed 2017-09-18 by jantje