SpenceKonde/ATTinyCore #169: Using operator new/delete or ArduinoSTL requires "cppfix.h"

bitrex, 2017-09-11:
Putting the following code in a file called "cppfix.h" in the directory: 

../.arduino15/packages/ATTinyCore/hardware/avr/1.1.4/cores/tiny/ 

and #including it at the top of a sketch allows operators new/delete and code using features from the ArduinoSTL.h library to compile OK! (Arduino 1.8.3)

```
#ifndef CPPFIX_H
#define CPPFIX_H

#include <stdlib.h>

__extension__ typedef int __guard __attribute__((mode (__DI__)));

void * operator new(size_t size);
void operator delete(void * ptr);

int __cxa_guard_acquire(__guard *g) {return !*(char *)(g);};
void __cxa_guard_release (__guard *g) {*(char *)g = 1;};
void __cxa_guard_abort (__guard *) {};

void * operator new(size_t size)
{
return malloc(size);
}

void operator delete(void * ptr)
{
free(ptr);
}

#endif //CPPFIX_H
```

SpenceKonde, 2017-09-11:
Hmm, I had never heard of that library; can you give a bit more background on this? 

It would be ideal if you could submit a PR for this - though I'd like to understand what exactly the function of this is before adding it to the core.  (note that it needs to be done in both tiny (for x5, x4, x61, x8, x7) and tinymodern (for x41, 1634 and 828)

bitrex, 2017-09-11:
Sure, the following example code compiles and runs OK for the Arduino/Genuino Uno, etc.
```
struct Foo 
{
  int bar;
  int baz;
  int buz; 
};

auto foo = new Foo{1, 2, 3};

void setup() {
  Serial.begin(9600);
  foo->bar = 3;
  foo->baz = 4;
  foo->buz = 5;
}

void loop() {
  Serial.print(foo->bar);
}
```
However when compiled for the ATTinyCore produces the following linker error:

> /tmp/ccVIJYFT.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_sketch_sep11b.ino.cpp.o.1790':
> ccVIJYFT.ltrans0.o:(.text.startup+0x3a): undefined reference to `operator new(unsigned int)'
> collect2: error: ld returned 1 exit status
> exit status 1
> Error compiling for board ATtiny25/45/85.

For a long time it seems the master Arduino core didn't include any implementations of the new/delete operators, the rationale probably being dynamically allocating/freeing memory is a Bad Idea on resource-limited embedded processors. But AVRs aren't my granpa's 8 bit; the Arduino API is C++ and currently supports C++11 features, and there are often good reasons to at least be able to use the various "new" operators sometimes. Like the example code above if I were to just declare `Foo foo; ` the dynamic fields for that class or struct would end up on the stack, which isn't always what you want. But then one also has to have definitions for "delete" even if that operator is never explicitly called, cuz the linker will complain about them not being available for use by say implicitly-declared destructors of POD structs and classes.

So it looks like at some point the master branch of Arduino included them due to popular demand since the first example seems to compile OK with no further #includes.

[ArduinoSTL](https://github.com/mike-matera/ArduinoSTL) a pretty complete port of uCLibc++ for the Arduino platform which includes versions of a bunch of standard template library functions like std::vector. The overhead for most of them isn't terrible, probably not feasible for say the Tiny25, but I mostly use the ATTiny85 and vectors and some of the other templates work just fine on that device, and LTO reduces the code size footprint even further. I wrote a little custom allocator amenable to uPs can be "plugged in" to std::vector and std::string's "allocator" template so that I can have vectors and strings that can be dynamically resized, exploit stack optimization, etc. without explicitly calling new and delete during those operations (there's a fixed-size static buffer allocated at startup) and getting fragmentation.

Unfortunately if I just drop the ArduinoSTL at the top of a sketch it gives the following linker error:

> /tmp/ccOcgZuj.ltrans0.ltrans.o: In function `std::basic_streambuf<char, std::char_traits<char> >::~basic_streambuf()':
> ccOcgZuj.ltrans0.o:(.text+0x496): undefined reference to `operator delete(void*)'
> /tmp/ccOcgZuj.ltrans0.ltrans.o: In function `std::basic_iserialstream<char, std::char_traits<char>, Stream>::~basic_iserialstream()':
> ccOcgZuj.ltrans0.o:(.text+0x4f6): undefined reference to `operator delete(void*)'
> /tmp/ccOcgZuj.ltrans0.ltrans.o: In function `std::basic_istream<char, std::char_traits<char> >::~basic_istream()':
> ccOcgZuj.ltrans0.o:(.text+0x50e): undefined reference to `operator delete(void*)'
> /tmp/ccOcgZuj.ltrans0.ltrans.o: In function `std::basic_serialbuf<char, std::char_traits<char>, Stream>::~basic_serialbuf()':
> ccOcgZuj.ltrans0.o:(.text+0x522): undefined reference to `operator delete(void*)'
> /tmp/ccOcgZuj.ltrans0.ltrans.o: In function `std::basic_oserialstream<char, std::char_traits<char>, Stream>::~basic_oserialstream()':
> ccOcgZuj.ltrans0.o:(.text+0x582): undefined reference to `operator delete(void*)'
> /tmp/ccOcgZuj.ltrans0.ltrans.o:ccOcgZuj.ltrans0.o:(.text+0x59a): more undefined references to `operator delete(void*)' follow
> collect2: error: ld returned 1 exit status
> exit status 1
> Error compiling for board ATtiny25/45/85

Ok, so I need these operators defined to even link this library, and currently ATTinyCore doesn't seem to define them. But if I just try to add "cppfix.h" to my own sketch directory I get _compiler_ errors that the operators have multiple definitions, i.e. the compiler can't figure out which definition it's supposed to be using, because ArduinoSTL has its own definitions somewhere to support say the C++ standard library "#include <new>". Adding "cppfix.h" to the core library folder seems to rectify this, I guess the compiler crunches all that before it looks at the libraries included at the top of the sketch, and everything compiles and links fine as it does with the current Arduino main core.

SpenceKonde, 2017-09-11:
Does putting 
https://github.com/arduino/Arduino/blob/2bfe164b9a5835e8cb6e194b928538a9093be333/hardware/arduino/avr/cores/arduino/new.h
and 
https://github.com/arduino/Arduino/blob/2bfe164b9a5835e8cb6e194b928538a9093be333/hardware/arduino/avr/cores/arduino/new.cpp

into
../.arduino15/packages/ATTinyCore/hardware/avr/1.1.4/cores/tiny/

instead of your ccpfix.h fix the problem? 

If so, i think that's the solution - keep it in sync with the main line core.

gion86, 2017-10-14:
Yes it does compile (and run correcty) for an tiny84 I've just tested with new.h/cpp.
At least in my fork https://github.com/gion86/ATTinyCore/tree/newduino which is a minimal version but  not really compatible with yours for a PR.
I too think to keep the Arduino line is the best solution.

SpenceKonde, 2017-10-25:
Should be fixed in github version now, will be in 1.1.5 board manager release.

commit 5a263293 (2017-10-25): Add new.h from latest arduino core

Fixes #169
closed 2017-10-25 by SpenceKonde