SpenceKonde/ATTinyCore #172: Free RX pin if Serial is used only for debgugging

gion86, 2017-10-16:
I'm using the Serial instance only for debugging, so the RX PIN is 'busy' for the serial but I would like to use it for other purpose. If I set the pinmode to output **AFTER** the `begin()` call the pin works as an output, but I a more elegant solution would be needed.. ;)
Is anyone interested in this? Or am I the first one to think about that??

SpenceKonde, 2017-10-16:
Which part is this?
Depending on which part, "Serial" can refer to one of three implementations, each with different behavior here.

gion86, 2017-10-16:
Ops sorry. It's a tiny84, I'm using TinySoftwareSerial implementation.

SpenceKonde, 2017-10-16:
This is functioning as designed (though your solution to changing the pin to work as an output is incomplete; you should also be disabling the analog comparator interrupt); changing to doc; I will document the appropriate way to disable RX and TX functionality on the various serial implementations. 

It is important to remember that using hardware serial functionality, you need to directly write registers to disable TX or RX as well; the Arduino Serial class does not include a wrapper for setting/unsetting the TXEN/RXEN bits of UCSRnB...

gion86, 2017-10-16:
Your right about the analog comparator, and yes let's follow the Arduino direction.
Thank you for the doc, always welcome.

closed 2018-01-26 by SpenceKonde