SpenceKonde/ATTinyCore #174: Add support for ATtiny212/412  and ATtiny214/414/814 and ATtiny416/816

Sembiance, 2017-10-25:
There are a bunch of new ATtiny chips coming soon:

ATtiny212/412: http://ww1.microchip.com/downloads/en/DeviceDoc/40001911A.pdf
ATtiny214/414/814: http://ww1.microchip.com/downloads/en/DeviceDoc/40001912A.pdf
ATtiny416/816: http://ww1.microchip.com/downloads/en/DeviceDoc/40001913A.pdf

These are quite exciting as they each have:
* I2C (Master+Slave)
* SPI (Master+Slave)
* USART
* DAC
* RTC
* Sleep Modes
* PTC (814 and 816 only)

The only one available for purchase now is the ATtiny414 from DigiKey: https://www.digikey.com/product-detail/en/microchip-technology/ATTINY414-SSFR/ATTINY414-SSFRCT-ND/7682217

I've got a couple of ATtiny414's on the way, but  sadly I won't be able to do any experimentation with them for a few months as I'm working on some other projects.

Just thought I'd get the ball rolling on adding support for these. I'm particularly excited about the ATtiny412 as it's a tiny 8-pin chip but still with all the features mentioned above.

SpenceKonde, 2017-10-25:
These are very exciting parts - but the process of supporting them in Arduino is a monumental task as every single library function must be rewritten. The peripherals (even GPIO) are pic-like, not AVR-like, so they need their own implementation of *EVERYTHING* - serial, digitalread/etc, wire, spi, pwm, adc input. 

This is a huge task and would be covered by a separate core. I do not expect to have time to do this - I certainly cannot do it alone. 

When you consider the amount of development effort involved, I legitimately question whether adding support for these parts makes sense - you can't leverage the existing arduino libraries any more than you could with a totally different architecture. And at that point, aren't there ARM chips that are more attractive?

closed 2017-10-25 by SpenceKonde

Sembiance, 2017-10-25:
Ahh, thanks for the info! Still somewhat new to arduino/electronics and wasn't aware they were so different underneath. Thanks.

SpenceKonde, 2017-10-25:
Yeah - it's a shame that they apparently made ZERO effort to maintain compatibility with AVR-land. 

I realize, probably needs incompatible implementation of all the new whizz-bang stuff - but the capabilities of Serial, I2C, SPI are essentially unchanged from the AVR line, as if - of course - GPIO. But alas, it's all different.

Sembiance, 2017-10-27:
I just wanted to thank you again for your helpful comments.

I decided that it's about time I moved away from Arduino and learned how to program a PIC directly with C. I'm very excited to learn all about this and am excited to be able to have a vast number of different MCU's available so I can choose the right one for the job!

So thanks again!

aabmar, 2019-04-03:
The good thing with these chips are price. You get a decent MCU for $0.30, so it's PIC price for an AVR.

SpenceKonde, 2019-04-04:
#288 is the new issue for this discussion.

SpenceKonde, 2019-06-21:
https://github.com/SpenceKonde/megaTinyCore This is happening now. Still not quite ready, but progressing quickly, follow that repo for updates