SpenceKonde/ATTinyCore #178: How to use the PCINT1 on ATtiny1634

alessio31183, 2017-11-18:
Dear all,
I'm trying to use the PCINT1 on ATtiny1634 with no success for now.
What I've done is to enable the interrupt in the setup:

  SREG |= (1 << SREG_I);    // Enable global interrupts 
  GIMSK |=  (1<<PCIE1);    // Enable pin change interrupt for PCINT1
  PCMSK0 =  (1<<PCINT1);          // Enable pin change interrupt for PA1 (Pin 7, RDY)

and then to define the interrupt function as:

ISR(PCINT1_ISR)       // interrupt service routine called when PCINT1 changes state 
{
....
}

what's wrong?

Thanks

alessio31183, 2017-11-21:
No sorry I've got problem but solved later on.
I had written the issue before to solve it.
You can close it.

Thanks

SpenceKonde, 2017-11-22:
Great, glad to hear it's working for you :-)

closed 2017-11-22 by SpenceKonde

colesnicov, 2018-05-31:
@alessio31183  Do you want to share a resume?