SpenceKonde/ATTinyCore #180: `SPI.transfer(buffer, size)` doesn't work

exzombie, 2017-12-07:
Hi!

I've been testing SPI on ATtiny84 with a LoRa module. [The docs for the SPI library](https://www.arduino.cc/en/Reference/SPI) state that `transfer(buffer, size)` overwrites `buffer` with received data which doesn't happen here. Or rather, the first byte it sends works, as far as I can see with a logic analyzer. After the first byte and on subsequent transfers, DO seemingly changes polarity and SCK remains low. The data in `buffer` is untouched.

SpenceKonde, 2017-12-10:
Thanks for reporting this.

SpenceKonde, 2017-12-10:
To expedite fix, please post code that reproduces this issue, but which works when compiled for something with real SPI instead of a USI.

exzombie, 2017-12-10:
I would, but I don't have a 3V device with hardware SPI available. I'll try to obtain one soonish to confirm what the correct behaviour is. In the meantime, I'll take a look at the code, with an eye towards #134; no promises though :) as my free time comes in short bursts.

exzombie, 2017-12-13:
Still waiting for a 3V device. But I've looked at the code and decided that I won't try to debug it. Instead, I'll rewrite it because I also found that the data order setting is only honored for `transfer(uint8_t)`; I'll fix that to match the hardware SPI version, i.e. this setting should change both bit and byte order.
I've looked at #167 where you expressed your concern about code size. So yeah, a version of the USI clocking code has to be generated for each clockdiv provided by the API. I can put it in the `.cpp` file, where it will always be available; it will increase code size, but LTO should remove unused code. Or, I can put it in the `.h` file, where only the code actually used will be instantiated, LTO or not. What do you think?

PR #184 (2017-12-18): New features in tinySPI
This PR fixes #180 and #134, implements
  * USI clocking with dividers 2, 4, 8 and >=14
  * SPI modes 2 and 3

Code size has increased, but mostly on account of fixes because previously, the `transfer` routines did not properly reverse data in the LSBFIRST case. Compiling a simple sketch that uses `SPISettings` and all three `transfer` functions, code size increases by:
  * 302 bytes for fixes, new modes and clock selection support
  * 24 bytes for clock divider >=14
  * 84 bytes for divider 2
  * 38 bytes for divider 4
  * 138 bytes for divider 8
  * 556 bytes when clo [... cut]
commit 667f65cb (2017-12-18): Merge pull request #184 from exzombie/master

Fix #180, #134 - Much thanks for your help on this!
closed 2017-12-18 by SpenceKonde