SpenceKonde/ATTinyCore #183: missing some operator

aster94, 2017-12-12:
hello dr Azzy,

i think that in your core there are "something" missing. the quotation marks are there because to  be honest i am not sure about what these things are

please see: https://forum.arduino.cc/index.php?topic=351930.0
i am using this library: https://github.com/0xPIT/encoder 
and it uses the "new" operator, it was missing so i added it with:

void *operator new(size_t size) {
  return malloc(size);
}

and it worked
i thought you would like to add these

SpenceKonde, 2017-12-13:
This is a duplicate of #169. It is fixed in the github version, but not yet in the board manager version.

closed 2017-12-13 by SpenceKonde

aster94, 2017-12-13:
Great to know that it is already fixed!