SpenceKonde/ATTinyCore #190: Support for the newer(?) 16x series (1614, 1616, 1617)?

toddzebert, 2018-01-24:
I've had a lot of success using this lib on the ATtiny85's, and now have moved up to these 16x's.

http://www.microchip.com/wwwproducts/en/ATTINY1614

Thanks!

SpenceKonde, 2018-01-24:
Duplicate. Closed. https://github.com/SpenceKonde/ATTinyCore/issues/174
I am getting REALLY sick of explaining to people, over and over, that I do not have time to rewrite EVERY SINGLE FUNCTION OF THE CORE to support the new PIC-like ATTinys. EVERYTHING is different, they might as well be a different architecture. 

If you want to work on this project, I'd be happy to work with you to get them added, but there's just a huge amount of work to do to reimplement all of the standard arduino libraries, and I have my hands full with my current commitments.

closed 2018-01-24 by SpenceKonde

toddzebert, 2018-01-24:
OK, got it. That issue didn't come up when I searched on variations of these models. Again, thanks for all your effort.

SpenceKonde, 2019-06-21:
https://github.com/SpenceKonde/megaTinyCore This is happening now. Still not quite ready, but progressing quickly, follow that repo for updates