SpenceKonde/ATTinyCore #198: digitalWriteFast on attiny1634

alessio31183, 2018-03-08:
Dear all,
I'm trying to make the porting for the digitalWriteFast library on the attiny1634.
I've just added this in the digitalWriteFast.h:

// --- AVR tiny uC ---
#elif defined(__AVR_ATtiny1634__)

#define UART_RX_PIN     (1) //PA7
#define UART_TX_PIN     (0) //PB0

#define I2C_SDA_PIN     (16) //PB1
#define I2C_SCL_PIN     (12) //PC1

#define SPI_HW_SS_PIN   (2)  //PA6
#define SPI_HW_MOSI_PIN (16) //PB1
#define SPI_HW_MISO_PIN (15) //PB2
#define SPI_HW_SCK_PIN  (12) //PC1

#define __digitalPinToPortReg(P) \
(((P) >= 1 && (P) <= 8) ? &PORTA : (((P) >= 9 && (P) <= 13) ? &PORTC : &PORTB))
#define __digitalPinToDDRReg(P) \
(((P) >= 1 && (P) <= 8) ? &DDRA : (((P) >= 9 && (P) <= 13) ? &DDRC : &DDRB))
#define __digitalPinToPINReg(P) \
(((P) >= 1 && (P) <= 8) ? &PINA : (((P) >= 9 && (P) <= 13) ? &PINC : &PINB))
#define __digitalPinToBit(P) \
(((P) >= 1 && (P) <= 8) ? (8-(P)) : (((P) >= 9 && (P) <= 10) ? (14-(P)) : (((P) >= 11 && (P) <= 13) ? (13-(P)) : (((P) >= 14 && (P) <= 16) ? (17-(P)) : (P)))))

But it seems not to work.
HERE 
[digitalWriteFast.zip](https://github.com/SpenceKonde/ATTinyCore/files/1792830/digitalWriteFast.zip)
my library used with examples.
The ToggleFast.ino doesn't work on the attiny1634.

Can anyone help me, please?

SpenceKonde, 2018-03-08:
I am not familiar with that library, but are you aware of the different handling of pullups in the 1634, 828, and 841/441? 

On normal AVR's, the pullup is controlled by the PORTx register when the DDRx register is set to make that pin an input. On these three tinys, the pullup is controlled by the PUEx (Pull Up Enable) register, independently of PORTx and DDRx registers. It doesn't look like the library currently takes account of that.

alessio31183, 2018-03-08:
It works adding the code above in the original digitalWriteFast.h

SpenceKonde, 2018-03-08:
Wait, which is the "original" library? one of the problems with github is the proliferation of forks, such that it's very hard to know what the "original" version of a library is

fastPinMode (which is part of that library) will 100% not work unless it is specifically adapted to these parts soas to use the PUE registers. 

When you say it "works" or "doesn't work", what does that mean? Is there a compiler error? unexpected behavior?

alessio31183, 2018-03-08:
In the previous post there is the link to the original one.
I tried the example ToggleFast in my modified version and comparated with the ToggleSlow.
I can see the difference using the oscilloscope where the Togglefast makes a 4MHz square signal on the pin selected in the example