SpenceKonde/ATTinyCore #199: Question about 1.1.5

beicnet, 2018-03-08:
Hi there,

Can't install it manually, tried all things, but seems not working!

Can you release the Boards Manager version?

Regards

SpenceKonde, 2018-03-08:
Board manager version is released now; it can take up to a day between the "release" being created and the board manager version being available through board manager due to steps involved in making the board manager json - that is done by @per1234 - and it needs to then be uploaded, which I couldn't do during the day because I didn't have the key file for SCP login on me. 

Manual install instructions are here, for reference: https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md

closed 2018-03-08 by SpenceKonde

beicnet, 2018-03-09:
@SpenceKonde Thank you, yes, I tried the manual installation, both method and did NOT work... :disappointed: