SpenceKonde/ATTinyCore #200: Can't get simple blink working...

gcormier, 2018-03-09:
I've got my fuses set at 8MHz, internal oscillator. I did find the bootloader requirement, but I'd already set them manually with avrdude. Did a bootloader burn for good measure.

I've got the blink sketch flashed, using PA0.

When the sketch is running, I can see the voltage going from 0.2-to 0.5 volts, with the proper timing - 1 second on, 1 second off.

I have a 0.1uF right next to the power pins....

A real head scratcher.

gcormier, 2018-03-09:
Just noticed

> You cannot use the Pxn notation (ie, PB2, PA1, etc) to refer to pins.

However, if I CTRL-CLICK this in vMicro, I can see PA0 = PORTA0 = 0.

sleemanj, 2018-03-09:
> I can see the voltage going from 0.2-to 0.5 volts

Sounds like the pin is not set to OUTPUT, in in INPUT mode if toggle between high and low it is only turning on/off the pullup

gcormier, 2018-03-09:
It is set to output. Here's the code

`void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(0, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(0, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(0, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
`

sleemanj, 2018-03-09:
Looks correct.  One assumes you are observing this problem with nothing connected to that pin to eliminate a hardware issue.

@SpenceKonde will want to know which chip, I don't think you have specified.

gcormier, 2018-03-09:
Sorry about that - ATTINY 841.

Yeah it's VCC, GND, a cap and a voltmeter.

SpenceKonde, 2018-03-09:
Pxn notation will not work. PA0 is defined by the files supplied with the compiler (the ones included by io.h) as the bit number of the bit corresponding to pin 0 on port A in the PORTA/PINA/PORTA registers. That means it's defined as 0. So is PB0, etc - so there's no way for the compiler to tell the difference between digitalWrite(PA0,HIGH), digitalWrite(PB0,HIGH), and digitalWrite(0,HIGH), 

Pin 0 is not PA0, please see the pin mapping image here: 
https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x41.md

You may refer to pins using PIN_xn (these are defines that point to the arduino pin number, supplied by this core), or using the arduino pin number, or the A# constants (these should be mapped to the digital pin numbers corresponding to it when used with digital functions.

gcormier, 2018-03-09:
That'll do it :)

closed 2018-03-09 by gcormier