SpenceKonde/ATTinyCore #202: Boards menu

MCUdude, 2018-03-09:
Is there an idea to reduce the number of available boards in the boards menu by moving the bootloader option inside the specific option? It would make the boards menu go from this:
  
[image]
  
.. To this:
[image]

I think this should be doable with some advanced binary magic.

SpenceKonde, 2018-03-09:
Sure - this is a low priority though.

sleemanj, 2018-03-09:
Example implementation in my distribution for reference... 

https://github.com/sleemanj/optiboot/blob/master/dists/diy_attiny/avr/boards.txt#L118

SpenceKonde, 2018-10-07:
I'm not sure how to do this, because we need the options on the bootloader menu to depend on which chip is selected.... (for example, we need the max upload size to depend on whether a bootloader is selected AND which size chip is selected.....

SpenceKonde, 2018-11-12:
Closed because doesn't appear to be possible (without a separate entry for each part instead of each family - which would offset any gains in terms of cleaning up the boards menu)

closed 2018-11-12 by SpenceKonde