SpenceKonde/ATTinyCore #203: Attiny841 device signature error

JonnyK05, 2018-03-20:
When trying to burn a bootloader or program I always got... 

avrdude: Device signature = 0x1e9315
avrdude: Expected signature for ATtiny841 is 1E 95 0F

Which is weird becasue that is the signature for a AtMega328.

To fix this I had to into my local avrdude.config file which it spits out when you try to program. Mine was at.
"C:\Users\JK\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5/avrdude.conf"

On line 1313 I changed "signature        = 0x1e 0x95 0x0f;" to "signature        = 0x1e 0x93 0x15;"

Once I made that change everything worked for the 841. 

I'm fairly new to programming so I apologize if this is an isolated thing.

SpenceKonde, 2018-03-20:
I am mystified as to how it could "work" after that. 

It sounds like you were using an arduino pro mini as ISP, and didn't disable reset (under some conditions, you need to disable autoreset on the arduino running the ArduinoAsISP sketch - I've never figured out what they are), and were talking to the bootloader on the '328 on the arduino pro mini you were using as ISP instead of the '841 you were trying to bootload. By changing the sig, you made it write the bootloader for the 841 to the pro mini - but I can't imagine how it "worked" after that...

sleemanj, 2018-03-20:
@SpenceKonde 

> On line 1313 I changed "signature = 0x1e 0x95 0x0f;" to "signature = 0x1e 0x93 0x15;"

Datasheet for Tiny814, page 222
> ATtiny841 0x1E 0x93 0x15

@JonnyK05  is correct I think :-)

sleemanj, 2018-03-20:
But looking at the ATTinyCore source, and the 1.1.5 release, that's already correct.  So... don't know why @JonnyK05 would have the wrong values to start with.

SpenceKonde, 2018-03-21:
Woah, okay. My bad. 

Also - do not confuse the attiny841 with the attiny814. The 814 is one of those new chips that is NOT supported by this core.

closed 2018-03-21 by SpenceKonde