SpenceKonde/ATTinyCore #204: Error compiling for board ATtiny25/45/85.

c-e-p-x-u-o, 2018-03-21:
Digispark with ATtiny85 works fine.

```
sketch/RGBSetDemo.ino.cpp.o: In function `ClocklessController<(unsigned char)1, 3, 4, 3, (EOrder)10, 0, false, 10>::showPixels(PixelController<(EOrder)10, 1, 4294967295ul>&)':
/root/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:144: undefined reference to `timer0_millis'
/root/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:144: undefined reference to `timer0_millis'
/root/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:144: undefined reference to `timer0_millis'
/root/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:144: undefined reference to `timer0_millis'
/root/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:144: undefined reference to `timer0_millis'
sketch/RGBSetDemo.ino.cpp.o:/root/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:144: more undefined references to `timer0_millis' follow
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATtiny25/45/85.
```

```
#include<FastLED.h>
#define NUM_LEDS 8

CRGBArray<NUM_LEDS> leds;

static uint8_t hue[NUM_LEDS];

void setup(){
  FastLED.addLeds<WS2811,1>(leds, NUM_LEDS);

  for(int i = 0; i < NUM_LEDS; i++){
    hue[i] = i;
  }
}

void loop(){ 
  for(int i = 0; i < NUM_LEDS; i++){   
    leds[i] = CHSV(hue[i],255,255);
    FastLED.delay(0);
    hue[i] ++;
  }
}
```

sleemanj, 2018-03-21:
http://forum.arduino.cc/index.php?topic=522364.0

Quoting from a post I made there

> Because FastLED makes some assumptions about the core, the following needed to be added to your sketch (at the top is fine)

    extern "C" {
      volatile unsigned long timer0_millis = 0; // FAKE FAKE FAKE!
    }

> because the ATTinyCore (and derived) don't use this variable name to track their millis, the only reason I saw that FastLED wants to access this is to "correct" millis counts when it is taking longer to operate and causing missed interrupts, so if you don't really care about millis accuracy it might be OK (might).

SpenceKonde, 2018-03-21:
What is done differently such that fastled works on the official core? ATTinyCore should not have compatibility breaking differences from the official core where they can be avoided.

c-e-p-x-u-o, 2018-03-21:
works with
```
extern "C" {
  volatile unsigned long timer0_millis = 0; // FAKE FAKE FAKE!
}
```
Thank you.

Could you add this extern to the core as likely done in digispark.

sleemanj, 2018-03-21:
Up to @SpenceKonde but I don't think it a good idea, the change I proposed is just a dummy one, it relies on the fact that you **probably** don't need accurate millis, perhaps you do.  I'm of the opinion that this is something that FastLED should fix, not ATTInyCore and until they do it's better you get at least an "easily fixed" error and know about it than get silently drifting millis.

> What is done differently such that fastled works on the official core? 

The official core stores it's whole-millis count in `timer0_millis`
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/wiring.c#L39

While ATTInyCore uses `millis_timer_millis` - which makes sense since the timer may well not be `timer0`
https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tiny/wiring.c#L101

sleemanj, 2018-03-21:
FWIW, a quick search of FastLED repo...

https://github.com/FastLED/FastLED/blob/f6c4e51ac7ce7ef7813eda91bb70dca5a3827634/platforms/avr/led_sysdefs_avr.h#L46

this was added 2 months ago according to the blame, so I think this is already fixed in FastLED then.

c-e-p-x-u-o, 2018-03-21:
Great, thank you!

SpenceKonde, 2018-03-21:
Oh wonderful - thanks James! 

I was about to go adding the #define to my core - but it sounds like they figured out what to do. 
So now, I can reply to this issue like this - one of my favorite issue resolutions:
@c-e-p-x-u-o this error is caused by a bug in the version of FastLED you are using - the FastLED team has already corrected this issue. Please update to the latest version of that library and let me know if you have any further issues.

closed 2018-03-21 by SpenceKonde