SpenceKonde/ATTinyCore #206: Attiny861 USI [Question]

klerone, 2018-04-10:
Dear.

I was reading the datasheet of the Attiny861 and I see it has kind of double USI interface, one in PORTB and other in PORTA. the one in PORTB is the one can be used to upload the code via ISP while the other seems to be as master only.

I could not find in your examples a way to sent it and use an I2C and SPI device in the same project, example (read BMP085 and send the data via RF24).

This is the datasheet [http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2588-8-bit-AVR-Microcontrollers-tinyAVR-ATtiny261-ATtiny461-ATtiny861_Datasheet-Summary.pdf](url)

klerone, 2018-04-10:
wait a moment....
reading the pin definitions in pins_arduino.h I found a gem.
It is set in different pins and as I read in the datasheet, SPI is set to be use in the PORTB and I2C is set to be used in PORTA (Arduino SDA = 0, SCL =2). 

Have someone else actually use it?

SpenceKonde, 2018-04-14:
My guess would be that it doesn't work (in fact, I'd go so far to say that either SPI or I2C on the 861 is busted - afaik nobody's tested it, but I'm almost certain the USIPP register is never manipulated, so one of those can't work); sadly, the 861 doesn't get much love from people. It's a crying shame too - it's a great part, available in DIP, but with more pins than the tiny84, and a really spiffy timer, and the PLL so you can clock it at 16mhz without a crystal. Of course, I'm part of the problem - I always use Tiny841 or Tiny1634 for my own projects. 

I do think it would be possible to adapt the unified Wire/SPI libraries to use one set of pins for I2C and the other for SPI - however the code would have to be smart enough to reconfigure the USI for I2C or SPI with every operation. Someone (probably me) would need to take a look at whether this can be implemented sanely. 

Finally, the examples in this core are mediocre at best (I didn't create or modify any of them, nor did anyone else, so where there are examples, they're leftover from before the library they're from was incorporated) - there's an open issue for writing better ones. I need to do a promotion "write examples, test them, and submit a PR and I'll mail you free electronics shit".

commit 93eedb68 (2018-10-08): Fix #206

SDA/SCL pins now reflect the fact that USIPP is not adjusted.
closed 2018-10-08 by SpenceKonde