SpenceKonde/ATTinyCore #207: Add support for ATTiny43

SpenceKonde, 2018-04-18:
Support for the ATTiny43 has been requested. This is a pretty boring 4k flash ATTiny, but it has the added feature of an on-chip boost converter (just-add-inductor-and-caps) for running off a single alkaline cell and generating 3v for external devices, albeit at very limited current. 

Should just require a variant file and boards.txt entry, nothing special.

SpenceKonde, 2018-05-03:
Oh - and this does need a bit more work than just a variant defined, because it's timer1 is an 8-bit one (copy of timer0) - I don't think it will work out of the box to get PWM out of timer1 (nor tone() )

tedder, 2018-05-28:
I'm working on a board design for this, if you manage to get this going I'll send you one.

SpenceKonde, 2018-05-28:
That's great! I look forward to it. How soon? Maybe I won't have to make one myself to test this! 

If you'd like to talk about the design of the board, have a second pair of eyes look over it before you sent it to the fab house, I'd be happy to take a look. email me spencekonde@gmail.com.

tedder, 2018-05-28:
I'll drop it your way. Hoping to order parts and send the boards off this
week, i'll be gone until ~jun10 so there'll be time for the parts to arrive.

tedder, 2018-05-29:
[you've got mail] in case you didn't see it.

SpenceKonde, 2018-09-24:
Quick note during review here - this will require routines to deal with the lameass 8-bit timer1. This may also be usable for the attiny26.

tedder, 2018-09-24:
interesting that you brought this up, I just ordered a new board for this.

SpenceKonde, 2018-09-24:
Don't get too excited, it doesn't mean I was working on it ;-) - I just looked at this and was like "wait, does that thing have a normal timer1?" - no, no it does not.

tedder, 2018-09-24:
Oh, no worries. I want to get a board working with a hello world LED in AVR and then I'll be anxious. I'll send you a populated board once I know it works.

SpenceKonde, 2018-11-05:
ToDo list here:
* Create pins_arduino,h for tiny43 variant
* Add support for adjusting CKDIV in init() for 4MHz operation off internal boost converter
* Implement PWM on 8-bit timer1
* Implement Tone on 8-bit timer1?

SpenceKonde, 2018-11-29:
Preliminary support is in. It *may* be totally working. 

PWM, Tone, SPI, and I2C need to be tested.

SpenceKonde, 2018-12-04:
If problems, please create new issue. Closing to clean up issues.

closed 2018-12-04 by SpenceKonde