SpenceKonde/ATTinyCore #208: i2c slave bug in USIWire

dontsovcmc, 2018-04-20:
The pull request hasn't merged, so I think, if somebody have problems with Attiny85 as i2c Slave go here: https://github.com/puuu/USIWire/issues/5

SpenceKonde, 2018-04-27:
Sorry for slow reply - I just saw this. There is no open pull request in this repo, so I'm not sure to what you're referring to. Do you have a version of the code that fixes a bug with I2C slave functionality on USI devices? If so, I would love to see a PR, or more details! I've been told of a bug in the slave functionality of USI wire, but i've never known what needed to be done to fix it, or whether the version of USIWire that went into my core contained the fix. The slave I2C code has not been well exercised (maybe not at all), so it's entirely possible that there's an issue I'm not aware of.

dontsovcmc, 2018-04-27:
This is pull request: https://github.com/puuu/USIWire/pull/6/files
The bug is rare: only if device has both i2c master and slave role.

closed 2018-04-27 by SpenceKonde