SpenceKonde/ATTinyCore #209: ATtiny88 I2C Wire library doesn't work correctly as slave writer

technoblogy, 2018-04-26:
Code demonstrating the problem (based on the Arduino Master Reader/Slave Writer example):

On Attiny88 (8MHz internal clock):

    #include <Wire.h>

    const int I2CAddress = 0x3A;

    void requestEvent() {
      Wire.write(0x12);
      Wire.write(0x34);
    }

    void setup() {
      Wire.begin(I2CAddress);
      Wire.onRequest(requestEvent);
    }

    void loop() {}

On Arduino Uno:

    #include <Wire.h>

    const int I2CAddress = 0x3A;

    void setup() {
      Wire.begin();
      Serial.begin(9600);
    }

    void loop() {
      Wire.requestFrom(I2CAddress, 2);

      while (Wire.available()) {
        int c = Wire.read();
        Serial.print(c, HEX); Serial.write(' ');
      }
      Serial.println();
      delay(5000);
    }

The Serial Monitor on the Uno prints: `34 FF`.

I tested the routines with two Arduino Unos and it prints `12 34` as expected.

SpenceKonde, 2018-04-27:
What's really weird here is that the Tiny88 actually has full I2C hardware; it's unclear to me why we're getting different results from the Uno, since the active code (after the #ifdefs ) should be literally identical to the Wire library supplied with the official core for the Uno.

SpenceKonde, 2018-04-27:
Found the problem, should be fixed by https://github.com/SpenceKonde/ATTinyCore/commit/79bdf227aa292d4e1c2132551a126f3d262707df

SpenceKonde, 2018-04-27:
Reopen issue if this turns out not to fix it (if it does fix it, i'd appreciate your letting me know too via comment on this issue.

closed 2018-04-27 by SpenceKonde

SamWarmuth, 2019-04-17:
I saw this issue as I was integrating an ATTiny88 into a project, and can confirm that i2c slave write messages are working 👍

SpenceKonde, 2019-04-17:
Thanks for confirmation