SpenceKonde/ATTinyCore #210: AT85 bootloader

DaveCalaway, 2018-05-01:
Hi everyone, 
i write to you for help me in a process ( right or not ) to upload the right bootloader on my AT85.

Usually i use my Digispark ( base on AT85 ) for my project, but i would like to use the IC directly.
According with [Digispark doc](https://goo.gl/sS3xtQ) they use the [micronucleus bootloader](https://goo.gl/cHh4Ki).

I used this [guide](https://goo.gl/xQzUHp) for upload the bootloader, using first the Arduino for ISP and in the end this command for uploade the boot:
`avrdude -C ../etc/avrdude.conf -c arduino -p t85 -P <Serial Port> -b 19200 -U flash:w:<micronucleus Directory>/firmware/releases/t85_default.hex
`
I burned the fuse according this [step](https://goo.gl/bUizgz).

Now i have connected the AT85 at my pc by a "[programmer](https://goo.gl/pbcCDm)" , but after a different tests with the configuration menu i can not upload the code on the board.

Does my steps are correct? Someone can help me?

CraigMarkwardt, 2018-05-01:
If you are programming the IC directly using Arduino-as-ISP, then why do you need a bootloader?  Just use the Arduino IDE to upload your sketch directly.  The bootloader is needed if you are going to program over USB serial, but you are programming the chip directly.

DaveCalaway, 2018-05-01:
@CraigMarkwardt 
I want to upload the sketch by USB using the programmer. But when i connect my AT85 on the Pc it's not read the usb port in Arduino.

I believed  that the problem was the bootloader and i tried to update it by micronucleus.

After that the Arduino IDE still not see the programmer....

CraigMarkwardt, 2018-05-01:
Arduino-as-ISP is a viable way to upload sketches to a Tiny.

If you do want to follow the micronucleus way to upload, that's fine but there are some additional steps, like setting fuses, etc. Also, need to make sure the IDE is properly configured as an Arduino board, using the micronucleus uploader program.

As a side note, when I first started with Adafruit's "Trinket" (ATTiny85), I had no luck until I realized my USB cable was power only.  That doesn't seem to be your problem, but connections are always something to check.  Connecting an ATTiny85 to USB is not 100% trivial.

And the micronucleus technique will rely on triggering a reset and connecting the IDE within 10 seconds to do the upload before the upload phase times out.

Sounds like kind of a big pain if you already have Arduino-as-ISP working, but do whatever you need to do.

DaveCalaway, 2018-05-01:
The cable is very crucial, I had the same problem with esp. The cable works fine.

Good idea, I can try to update the bootloader of the IC gemma from Adafruit. 

I will write a review if all will go right.

DaveCalaway, 2018-05-01:
Thx you in any way for your help.

The Arduino-as-ISP require the Arduino for load the sketch, is it right?

I don't like to use micronucleus, it was in the first guide but is doesn't works correctly for me.
I chose the Adafruit bootloader because the USBtinyISP works without the IDE seen the port.
The process did you describe need the port, right?

According to the main page, only ATtiny 841, 828, 1634, 87, 167 have a bootloader in this package.

SpenceKonde, 2018-05-01:
This core does not provide support for Digispark boards at this time - you can program them via ISP like any other attiny85, but the USB upload functionality will be lost - for this connect any supported ISP programmer, do burn bootloader, and upload sketches normally. 

I will investigate whether support for the digispark bootloader can be added to the core, as we have had a great many inquiries about supporting it recently.

tomaskovacik, 2019-01-13:
hello everyone!

I do made it, sort of

here it is: https://github.com/tomaskovacik/ATTinyCore/commit/b8e6e4a35b0fbac0c97a807221e42f70bcb83596

issues: 
digispark is *it ... 16.5MHz speed while connected to USB, powered separately it is 16MHz (sort of, 10% plus minus ... =>https://digistump.com/wiki/digispark/tricks, and I can say it;s problematic, I use timer tick at 100us and it was not reliable ... 

micronucleus need to be installed https://github.com/micronucleus/micronucleus/tree/master/commandline as tool in arduino IDE. I did not do this, I simply install https://github.com/digistump/DigistumpArduino which install micronucleus (2.0a4 => buggy, replace binary with one from  micronucleus, but when bootloader need to be upgraded on digispark:  https://kovo-blog.blogspot.com/2019/01/how-to-upgrade-bootloader-on-digistump.html

digistump guys use some sort of launcher not directly micronucleus, so this need to be investigated if it need to be there but my guess: this launcher is used just to support more products from digistump

for now I'm done investigating this, cose I brick my digispark :) so I need build  HV programer

thx for great core! 

t.

SpenceKonde, 2019-01-13:
Thanks for the work. It sounds like the land of digispark is kind of a shitshow currently, what with boards coming with bad bootloader preinstalled and the problem with the upload tool? Don't some of the boards come with RSTDSBL set so you can't easily update the bootloader without an HV programmer? We'd need a way to make board manager pull in a good copy of the microneucleus tool (since you say the one that the package comes with is no good), including a working one for windows, mac, and linux.... I'm not sure how to achieve that. 

I've got two digispark boards coming in the mail.

tedder, 2019-01-13:
(note digispark is also about two years late on a 3d printer kickstarter project too..)

per1234, 2019-01-14:
I don't think Digistump has sold any boards in quite a while. It's likely all just clones now.

Their hardware package doesn't seem to be actively maintained anymore. I think it could be really useful to have an alternate support option for these popular boards.

For Boards Manager, the only thing needed is to host Micronucleus somewhere. DigiStump just put their copies of it on GitHub and I didn't hear any reports of problems with that.

tomaskovacik, 2019-01-15:
launcher is just rename avr-dummy: https://github.com/digistump/DigistumpArduino/issues/15

probably to be "arduino compatible", explained:

using jus  micronucleus cmd binnary work with only problem and it is, nothing is showed in status windows of arduino ide, so this imporatant msg is not printed:

> Please plug in the device ... 
> Press CTRL+C to terminate the program.

for this probably digistum create dumy-avrdude (my guess),  but as this is not important binarty it can by taken from digistump package and just "use it again", no need to compile it ... 

here is my creation: https://github.com/tomaskovacik/ATTinyCore/tree/Digistump

what is done:
digispark and pro support (in menu is attiny45/85 (micronucleus,Digispark), similar with digispark pro)
compile and upload fine
bootloader upgrade works fine

what need to be checked:
whole attiny45 and digispark Pro  support is just theoretical (on my side, I did not have any chips on hand, also support is questionable on attiny45, 4k space and almost half will be bootloader)
other options in board menu are not tested:
LTO
Timer1 speed

digispark use PB5 as IO, so this need to be probably somehow fixed in pin definition 

json file with support is here, if anyone is willing to try it :)

http://temp.nodomain.sk/package_drazzy.com_index.json

or just archives:
http://temp.nodomain.sk/ATTinyCore-vg1.2.3.tar.gz
http://temp.nodomain.sk/micronucleus-2.0a5-linux64.tar.gz

tomaskovacik, 2019-01-18:
ok, one problem, while using

#include <SoftwareSerial.h>
SoftwareSerial mySerial(PB4, PB3); 

timing incoming data did not work using INT0 and TIMER1

just saying

but "only output" serial lib TinyDebugSerial works.

SpenceKonde, 2020-06-24:
digispark bootloader support will be in 1.4.0, see #414

closed 2020-06-24 by SpenceKonde