SpenceKonde/ATTinyCore #211: Which Timers Used

CraigMarkwardt, 2018-05-01:
Hi, great project, thanks!  Your documentation is really nice, but what I'd like to know is which timers are being used.  In this case, I would like to use Timer1 on ATTiny85 (which I am aware is an odd duck), but it would be nice to have documented which Timers are occupied by the core.  Thanks.

SpenceKonde, 2018-05-01:
All of the ATTinyCore boards should use Timer0 for millis and Timer1 for Tone (ATTiny841/441 should probably be changed to use Timer2, so that Tone can be used on them without interfering with other libraries that use Timer1 - The 841 has two identical 16-bit timers) 

You can freely use Timer1 on the ATTiny85 - out of the box it is configured for PWM as shown on the pinout chart, but that's all that's done to Timer1 on the Tiny85. I agree that Timer1 on the attiny85 is one of it's most interesting features, and I'm happy to say that you can use it without trashing millis()/delay()/etc.

SpenceKonde, 2018-05-03:
Moved tone() to timer2 on the 841/441 while I was looking at this, so that you can use Servo/TimerOne/etc at the same time as tone() on those devices.

closed 2018-05-03 by SpenceKonde