SpenceKonde/ATTinyCore #212: twi.c fails to compile

mortie, 2018-05-03:
The file [twi.c](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/Wire/src/twi.c) fails to compile, complaining that `true` and `false` are undefined.

It seems like `Arduino.h` is supposed to include `<stdbool.h>`, or otherwise define true/false. The [default Arduino.h](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Arduino.h) includes `stdbool.h` at least.

SpenceKonde, 2018-05-03:
This regression was introduced by fix for #193

Floessie, 2018-05-07:
@mortie @SpenceKonde I'm really sorry that my clean up request caused you trouble. Anyway, it's nice to see that everything is up to date and fine now. Great work! :+1:

commit 8399140f (2018-05-03): fix #212 for tinymodern